react-native-camera/react-native-camera #1098: Erro after update to master

p-ugulino, 2018-01-12:
`Event.java:24: error: no suitable method found for init(int)
    event.init(viewTag);
         ^
    method Event.init(int,long) is not applicable
      (actual and formal argument lists differ in length)
    method FaceDetectionErrorEvent.init(int,RNFaceDetector) is not applicable
      (actual and formal argument lists differ in length)
/Users/Pablo/repos/cadastro-cliente-app/cadastro_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/FaceDetectionErrorEvent.java:29: error: incompatible types: RNFaceDetector cannot be converted to long
    super.init(viewTag, faceDetector);
                        ^
/Users/Pablo/repos/cadastro-cliente-app/cadastro_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/CameraMountErrorEvent.java:22: error: method init in class Event<T> cannot be applied to given types;
    event.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/Pablo/repos/cadastro-cliente-app/cadastro_app/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/FacesDetectedEvent.java:52: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
4 errors
:react-native-camera:compileReleaseJavaWithJavac FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-camera:compileReleaseJavaWithJavac'.
> Compilation failed; see the compiler error output for details.
`
Somehow im getting this issue. Any idea how to fix?
Everything was fine until I update the repo.

closed 2018-01-12 by p-ugulino

ywongweb, 2018-01-14:
@p-ugulino Bump into the same error, did you managed to find a fix?

marcusvbp, 2018-01-18:
same error here.

p-ugulino, 2018-01-18:
@ywongweb i dont actually recall the steps i had to do to get it fixed but I remember that I started updating everything lol

marcusvbp, 2018-01-18:
after a downgrade to 0.12.0 and a gradlew clean I can build the project again.