react-native-camera/react-native-camera #1100: camera.capture() does not return a promise

seegurkegr, 2018-01-15:
### Steps to reproduce
This is my code:
```
    takePicture() {
        console.log("Take Picture")
        this.setState({loading: true})
        const options = {}
        this.camera.capture({metadata: options})
            .then((data) => {
                // mostly never displayed 
                console.log("Ready")
                console.log(data.path)

                this.props.savePicturePath(data.path) // redux stuff
                this.setState({loading: false}) 

                // reset camera screen - tried without this navigation stuff - but got the same behaviour
                this.props.navigation.dispatch(resetAction)
                // navigate to "UploadScreen"
                this.props.navigation.navigate('upload')
            })
            .catch(err => console.error(err));
    }
```

### Expected behaviour
Should return a promise. 

### Actual behaviour
Sometimes working (maybe just 10% of attempts ?) - and sometimes not (without any changes in my code)
If it is not working, there is no promise. Not even an error message.

### Environment
- **Node.js version**: 6.12.3
- **React Native version**:
- **React Native platform + platform version**: Android 7.0

### react-native-camera
**Version**: npm version 0.12.0

seegurkegr, 2018-01-15:
Okay i think i found the problem:
I'm trying to show a Spinner while the System is capturing. That's why i'm setting the loading state inside the component. Dependent on this state i've got a different rendering for the "CamUI". Obviously this seems to be the problem. Interestingly, this problem does not always occur. 
Here's the other part of my code:
```
    state = {
            cameraType: Camera.constants.Type.front,
            loading: false
        }
   
    renderCamUI(){
        if(this.state.loading){
            return (
            <View>
                <Spinner size="large"/>
            </View>
            )
        }else{
            return(
                <View>
                <Icon
                    name="photo-camera"
                    onPress={this
                    .takePicture
                    .bind(this)}
                    size={40}/>
                <Icon
                    name="cached"
                    onPress={this
                    .changeCameraType
                    .bind(this)}
                    size={25}/>
            </View>
            )
        }
    }

    render() {
        return (
            <View style={styles.container}>
                <Camera
                    ref={(cam) => {
                    this.camera = cam;
                }}
                    style={styles.preview}
                    aspect={Camera.constants.Aspect.fill}
                    type={this.state.cameraType}>
                   
                {this.renderCamUI()}
                </Camera>
            </View>
        )
    }
```
Can anyone make me some suggestions how to show a nice Spinner while capturing ? 
Thanks !

somonek, 2018-01-22:
I had the same issue and now it works for me with rendering other elements inside. You probably need to add 
```
<key>NSPhotoLibraryAddUsageDescription</key>
<string>MyApp wants to access your photos.</string>
``` 
to your info.plist in the ios project. That solved the problem to me.
Note that I had `NSPhotoLibraryUsageDescription` previously and it wasn't enough, just in case you have the same situation.

closed 2018-01-22 by sibelius