react-native-camera/react-native-camera #1101: RNCamera.recordAsync not working

wassgha, 2018-01-15:
### Steps to reproduce
1. Implement `RNCamera.recordAsync` on the RNCameraExample
2. Run the example and try recording a video

### Expected behaviour
`RNCamera.recordAsync` should return a `Promise` that resolves as soon as `stopRecording` is called or the video duration is exceeded.

### Actual behaviour
`RNCamera.recordAsync` never resolves

### Environment
- **Node.js version**: 5.6.0
- **React Native version**: 0.52.0
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**: "master"

wassgha, 2018-01-15:
I can look more into it as soon as I have time.

sibelius, 2018-01-15:
some code are missing on android, you can copy them from here https://github.com/expo/expo/blob/master/android/expoview/src/main/java/versioned/host/exp/exponent/modules/api/components/camera/CameraModule.java#L222

wassgha, 2018-01-15:
There's actually a TODO on it that I just noticed

https://github.com/react-native-community/react-native-camera/blob/b8061b46f783528c07aa5507f870e24f58c2f1fc/android/src/main/java/org/reactnative/camera/CameraViewManager.java#L166

would be great if whoever made the TODO clarifies what's needed, the code seems to be analogous to that of Expo, not sure what's missing.

sibelius, 2018-01-15:
I think the missing part is only the permissions, just copy and paste some more code, and it will be good

sibelius, 2018-01-18:
we can just remove the permission part, as we already is checking on `js` code

wassgha, 2018-01-22:
Reopening since #1130 doesn't fix.

### Reproducing the bug

1. Set line 100 in `RNCameraExample/App.js` to
```js
      this.camera.recordAsync({
        mute: true // Not setting this causes another error...
```
2. Try to record a video
3. You'd get a promise rejection error (`Starting video recording failed. Another recording might be in progress`)

sibelius, 2018-01-23:
`mute: true // Not setting this causes another error...`

we need a PermissionAndroid check for android is this case

we can do this on `js` code

jgfidelis, 2018-01-23:
Just tested on master, the behavior right now is the promise is only resolved when you call stopRecording().

I ran the example, started recording and when I called stopRecording() I got the response:
{uri: "file:///data/user/0/com.rncameraexample/cache/Camera/b45f1abd-2a6f-4055-933f-c90bf1f1304e.mp4"}

Is this the expected behavior? Or would it be better to make stopRecording async (instead of startRecording) and make it resolve the promise?

closed 2018-02-08 by wassgha

airam666, 2020-12-03:
this is not fixed my dudes. you need to check up on this unless this project is abandoned

commit 504d4b1f (2018-01-21): feature(rncamera): enable record on RNCamera on android, fix #1101
commit d8c4e1f3 (2018-01-21): Merge pull request #1130 from react-native-community/feature/recordRNCamera

feature(rncamera): enable record on RNCamera on android, fix #1101