react-native-camera/react-native-camera #1102: FixOrientation needed on iPhone SE

miguelcanobbio, 2018-01-15:
While taking a picture on iPhone SE iOS 10.3.3 the picture seems to be fine, but really is rotated 90º to right, this fixes when use the prop fixOrientation but in readme you tag it as Deprecated.

In readme you specify that we notify you if this happens

jgcmarins, 2018-01-15:
Hello @mcanobbio, could you try it on `master branch`?
Follow this: https://github.com/react-native-community/react-native-camera#how-to-use-master-branch

mcanobbio, 2018-01-18:
I'll try it and notify you if works ;), thanks

mcanobbio, 2018-01-18:
Still need the FixOrientation prop to works correctly in iPhone SE, when i take the picture everything seems to be fine, cause I put the path in a Image component and looks perfect, but I send the picture to an OCR and the picture is rotated 90º to right, otherwise when I use the FixOrientation prop everything looks good in OCR.

In Android works perfectly without fixOrientation

tirrorex, 2018-01-22:
@mcanobbio it's because how pictures works on ios.
I assume you're taking a picture in portrait, if that's the case, you are actually taking a picture in landscape with a 90 degrees rotation in the metadata.
So depending on where you are sending the picture and what you do with it (for example redimensionning the picture thus creating a new one), the rotation might not be taking into account (because sometimes you can lose the metadata).
What FixOrientation does is removing the rotation from the metadata and actually rotate your image making it a true portrait picture.

Hope i was clear enough :)

mcanobbio, 2018-01-29:
So, the fixOrientation prop should be always activated for my case on iOS, cause I send the picture to Google Cloud Storage and everything looks good, and same Image that I uploaded to Storage send to Google Cloud Vision and the image is rotated 90º

matias91, 2018-03-21:
Same problem here on a Samsung Galaxy S8.
The captured image is rotated 90º, using master branch.
Any suggestions to fix this?

Thanks in advance!

matias91, 2018-04-02:
This worked for me:
https://github.com/react-native-community/react-native-camera/issues/581#issuecomment-374074936

SimonErm, 2018-09-14:
#1786 
#1574

closed 2018-09-14 by SimonErm