react-native-camera/react-native-camera #1104: android crash when open and close camera many times

pj0579, 2018-01-16:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:v6.3.1
- **React Native version**:0.50.3
- **React Native platform** :  Android 7.1.1,  MI 6

### react-native-camera
**Version**: 0.13.0

android.content.res.Resources$NotFoundException: Drawable android:drawable/seekbar_thumb_material_anim with resource ID #0x1080690
Caused by: java.lang.NullPointerException: Attempt to read from field 'android.animation.Animator android.animation.AnimatorSet$Node.mAnimation' on a null object reference
 at android.animation.AnimatorSet.clone(AnimatorSet.java:725)
 at android.animation.AnimatorSet.clone(AnimatorSet.java:682)
 at android.animation.Animator$AnimatorConstantState.newInstance(Animator.java:589)
 at android.animation.Animator$AnimatorConstantState.newInstance(Animator.java:588)
 at android.content.res.ConstantState.newInstance(ConstantState.java:53)
 at android.content.res.ConstantState.newInstance(ConstantState.java:61)
 at android.content.res.ConfigurationBoundResourceCache.getInstance(ConfigurationBoundResourceCache.java:40)
 at android.animation.AnimatorInflater.loadAnimator(AnimatorInflater.java:114)
 at android.graphics.drawable.AnimatedVectorDrawable$AnimatedVectorDrawableState$PendingAnimator.newInstance(AnimatedVectorDrawable.java:808)
 at android.graphics.drawable.AnimatedVectorDrawable$AnimatedVectorDrawableState.inflatePendingAnimators(AnimatedVectorDrawable.java:784)
 at android.graphics.drawable.AnimatedVectorDrawable.applyTheme(AnimatedVectorDrawable.java:613)
 at android.graphics.drawable.DrawableContainer$DrawableContainerState.applyTheme(DrawableContainer.java:953)
 at android.graphics.drawable.DrawableContainer.applyTheme(DrawableContainer.java:612)
 at android.graphics.drawable.StateListDrawable.applyTheme(StateListDrawable.java:371)
 at android.graphics.drawable.AnimatedStateListDrawable.applyTheme(AnimatedStateListDrawable.java:392)
 at android.content.res.ResourcesImpl.loadDrawable(ResourcesImpl.java:594)
 at android.content.res.MiuiResourcesImpl.loadDrawable(MiuiResourcesImpl.java:308)
 at android.content.res.Resources.loadDrawable(Resources.java:876)
 at android.content.res.TypedArray.getDrawable(TypedArray.java:930)
 at android.widget.AbsSeekBar.<init>(AbsSeekBar.java:99)
 at android.widget.SeekBar.<init>(SeekBar.java:85)
 at android.widget.SeekBar.<init>(SeekBar.java:81)
 at com.facebook.react.views.slider.ReactSlider.<init>(ReactSlider.java:56)
 at com.facebook.react.views.slider.ReactSliderManager.createViewInstance(ReactSliderManager.java:121)
 at com.facebook.react.views.slider.ReactSliderManager.createViewInstance(ReactSliderManager.java:39)
 at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:44)
 at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:224)
 at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:153)
 at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.dispatchPendingNonBatchedOperations(UIViewOperationQueue.java:1006)
 at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:977)
 at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
 at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:136)
 at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
 at android.view.Choreographer$CallbackRecord.run(Choreographer.java:874)
 at android.view.Choreographer.doCallbacks(Choreographer.java:688)
 at android.view.Choreographer.doFrame(Choreographer.java:620)
 at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:862)
 at android.os.Handler.handleCallback(Handler.java:754)
 at android.os.Handler.dispatchMessage(Handler.java:95)
 at android.os.Looper.loop(Looper.java:163)
 at android.app.ActivityThread.main(ActivityThread.java:6381)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:901)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:791)

jgcmarins, 2018-01-16:
this not seems to be related to the lib

pj0579, 2018-01-16:
when I use react-native-camera 0.13.0 RNCamera  It happened

closed 2019-06-06 by pj0579