react-native-camera/react-native-camera #1105: Fatal Exception: NSRangeException on didFinishRecordingToOutputFileAtURL

mike623, 2018-01-16:
### Steps to reproduce
1. turn on camera
2.recording
3. stop record, crash

### Expected behaviour
not crash, return video object manually 

### Actual behaviour
crash, 

````
Fatal Exception: NSRangeException*** -[__NSArrayM objectAtIndex:]: index 0 beyond bounds for empty array
--
````
on
````
-[RCTCameraManager captureOutput:didFinishRecordingToOutputFileAtURL:fromConnections:error:]
````

### Environment
- **Node.js version**:->      v6.11.2
"react-native": "0.46.4",
iOS 11

### react-native-camera
"react-native-camera": "^0.6.0",

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgcmarins, 2018-01-16:
Hey @mike623 could you try to update `react-native-camera` dep?
It's on 0.13.0 now.
Thanks.

mike623, 2018-01-16:
Will try thx. Btw, I use another app to re-open camera. It is fixed. I suspect the connection lost between react bridge

closed 2018-01-22 by mike623