react-native-camera/react-native-camera #1109: The master branch compiled failed on Android.

jaggerwang, 2018-01-17:
### Steps to reproduce
1. Change the version in package.json from '0.10.0' to 'git+https://git@github.com/react-native-community/react-native-camera'
2. Gradle sync in Android Studio

### Expected behaviour
Gradle sync success.

### Actual behaviour
Gradle sync failed. I'm already using v0.10.0 for a few months, but the qrcode detecting is not good enough, so I want to try the Expo one.
[image]

### Environment
- **Node.js version**: v6.11.3
- **React Native version**: 0.44.3
- **React Native platform + platform version**: Android Studio 3.0

### react-native-camera
**Version**: master

sibelius, 2018-01-17:
Check Q & A section on README

closed 2018-01-17 by sibelius

lynkxyz, 2018-01-18:
@sibelius si  I was doing but it not working