react-native-camera/react-native-camera #1110: Invariant Violation: Native component for "RNCamera" dose not exist

ShanZiJun, 2018-01-17:
### Steps to reproduce
1.react-native init trainManage 
2.Manual install  ios 
// !$*UTF8*$!
{
	archiveVersion = 1;
	classes = {
	};
	objectVersion = 46;
	objects = {

/* Begin PBXBuildFile section */
		00C302E51ABCBA2D00DB3ED1 /* libRCTActionSheet.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 00C302AC1ABCB8CE00DB3ED1 /* libRCTActionSheet.a */; };
		00C302E71ABCBA2D00DB3ED1 /* libRCTGeolocation.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 00C302BA1ABCB90400DB3ED1 /* libRCTGeolocation.a */; };
		00C302E81ABCBA2D00DB3ED1 /* libRCTImage.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 00C302C01ABCB91800DB3ED1 /* libRCTImage.a */; };
		00C302E91ABCBA2D00DB3ED1 /* libRCTNetwork.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 00C302DC1ABCB9D200DB3ED1 /* libRCTNetwork.a */; };
		00C302EA1ABCBA2D00DB3ED1 /* libRCTVibration.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 00C302E41ABCB9EE00DB3ED1 /* libRCTVibration.a */; };
		00E356F31AD99517003FC87E /* trainManageTests.m in Sources */ = {isa = PBXBuildFile; fileRef = 00E356F21AD99517003FC87E /* trainManageTests.m */; };
		133E29F31AD74F7200F7D852 /* libRCTLinking.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 78C398B91ACF4ADC00677621 /* libRCTLinking.a */; };
		139105C61AF99C1200B5F7CC /* libRCTSettings.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 139105C11AF99BAD00B5F7CC /* libRCTSettings.a */; };
		139FDEF61B0652A700C62182 /* libRCTWebSocket.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 139FDEF41B06529B00C62182 /* libRCTWebSocket.a */; };
		13B07FBC1A68108700A75B9A /* AppDelegate.m in Sources */ = {isa = PBXBuildFile; fileRef = 13B07FB01A68108700A75B9A /* AppDelegate.m */; };
		13B07FBD1A68108700A75B9A /* LaunchScreen.xib in Resources */ = {isa = PBXBuildFile; fileRef = 13B07FB11A68108700A75B9A /* LaunchScreen.xib */; };
		13B07FBF1A68108700A75B9A /* Images.xcassets in Resources */ = {isa = PBXBuildFile; fileRef = 13B07FB51A68108700A75B9A /* Images.xcassets */; };
		13B07FC11A68108700A75B9A /* main.m in Sources */ = {isa = PBXBuildFile; fileRef = 13B07FB71A68108700A75B9A /* main.m */; };
		140ED2AC1D01E1AD002B40FF /* libReact.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 146834041AC3E56700842450 /* libReact.a */; };
		146834051AC3E58100842450 /* libReact.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 146834041AC3E56700842450 /* libReact.a */; };
		2D02E4BC1E0B4A80006451C7 /* AppDelegate.m in Sources */ = {isa = PBXBuildFile; fileRef = 13B07FB01A68108700A75B9A /* AppDelegate.m */; };
		2D02E4BD1E0B4A84006451C7 /* Images.xcassets in Resources */ = {isa = PBXBuildFile; fileRef = 13B07FB51A68108700A75B9A /* Images.xcassets */; };
		2D02E4BF1E0B4AB3006451C7 /* main.m in Sources */ = {isa = PBXBuildFile; fileRef = 13B07FB71A68108700A75B9A /* main.m */; };
		2D02E4C21E0B4AEC006451C7 /* libRCTAnimation.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 5E9157351DD0AC6500FF2AA8 /* libRCTAnimation.a */; };
		2D02E4C31E0B4AEC006451C7 /* libRCTImage-tvOS.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 3DAD3E841DF850E9000B6D8A /* libRCTImage-tvOS.a */; };
		2D02E4C41E0B4AEC006451C7 /* libRCTLinking-tvOS.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 3DAD3E881DF850E9000B6D8A /* libRCTLinking-tvOS.a */; };
		2D02E4C51E0B4AEC006451C7 /* libRCTNetwork-tvOS.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 3DAD3E8C1DF850E9000B6D8A /* libRCTNetwork-tvOS.a */; };
		2D02E4C61E0B4AEC006451C7 /* libRCTSettings-tvOS.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 3DAD3E901DF850E9000B6D8A /* libRCTSettings-tvOS.a */; };
		2D02E4C71E0B4AEC006451C7 /* libRCTText-tvOS.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 3DAD3E941DF850E9000B6D8A /* libRCTText-tvOS.a */; };
		2D02E4C81E0B4AEC006451C7 /* libRCTWebSocket-tvOS.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 3DAD3E991DF850E9000B6D8A /* libRCTWebSocket-tvOS.a */; };
		2D16E6881FA4F8E400B85C8A /* libReact.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 2D16E6891FA4F8E400B85C8A /* libReact.a */; };
		2DCD954D1E0B4F2C00145EB5 /* trainManageTests.m in Sources */ = {isa = PBXBuildFile; fileRef = 00E356F21AD99517003FC87E /* trainManageTests.m */; };
		5E9157361DD0AC6A00FF2AA8 /* libRCTAnimation.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 5E9157331DD0AC6500FF2AA8 /* libRCTAnimation.a */; };
		832341BD1AAA6AB300B99B32 /* libRCTText.a in Frameworks */ = {isa = PBXBuildFile; fileRef = 832341B51AAA6A8300B99B32 /* libRCTText.a */; };
		A4B426C8200FCEE800431FF9 /* libRCTCamera.a in Frameworks */ = {isa = PBXBuildFile; fileRef = A4B42697200FCEB100431FF9 /* libRCTCamera.a */; };
		ADBDB9381DFEBF1600ED6528 /* libRCTBlob.a in Frameworks */ = {isa = PBXBuildFile; fileRef = ADBDB9271DFEBF0700ED6528 /* libRCTBlob.a */; };
/* End PBXBuildFile section */

/* Begin PBXContainerItemProxy section */
		00C302AB1ABCB8CE00DB3ED1 /* PBXContainerItemProxy */ = {
			isa = PBXContainerItemProxy;
			containerPortal = 00C302A71ABCB8CE00DB3ED1 /* RCTActionSheet.xcodeproj */;
			proxyType = 2;
			remoteGlobalIDString = 134814201AA4EA6300B7C361;
			remoteInfo = RCTActionSheet;
		};
		00C302B91ABCB90400DB3ED1 /* PBXContainerItemProxy */ = {
			isa = PBXContainerItemProxy;
			containerPortal = 00C302B51ABCB90400DB3ED1 /* RCTGeolocation.xcodeproj */;
			proxyType = 2;
			remoteGlobalIDString = 134814201AA4EA6300B7C361;
			remoteInfo = RCTGeolocation;
		};
		00C302BF1ABCB91800DB3ED1 /* PBXContainerItemProxy */ = {
			isa = PBXContainerItemProxy;
			containerPortal = 00C302BB1ABCB91800DB3ED1 /* RCTImage.xcodeproj */;
			proxyType = 2;
			remoteGlobalIDString = 58B5115D1A9E6B3D00147676;
			remoteInfo = RCTImage;
		};
		00C302DB1ABCB9D200DB3ED1 /* PBXContainerItemProxy */ = {
			isa = PBXContainerItemProxy;
			containerPortal = 00C302D31ABCB9D200DB3ED1 /* RCTNetwork.xcodeproj */;
[... 1417 lines cut ...]
/* End XCConfigurationList section */
	};
	rootObject = 83CBB9F71A601CBA00E9B192 /* Project object */;
}
[image]

3.react-native run ios

### Expected behaviour
run RNCameraExample 
### Actual behaviour

on the simulator 

Invariant Violation: Native component for "RNCamera" dose not exist

### Environment
- **Node.js version**: v9.4.0
- **React Native version**: 0.52.0
- **React Native platform + platform version**: iOS 11.2

### react-native-camera
**Version**: "react-native-camera": "^0.13.0"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sibelius, 2018-01-21:
fixed on master, can u try again?

closed 2018-01-21 by sibelius

adrianha, 2018-01-28:
Hi @sibelius , just tried using "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera" as dependency instead "react-native-camera": "0.13.0". Its not working too. it returned:
[image]
"react-native-camera": "0.13.0" is working on android.
Am i missing something?

sibelius, 2018-01-28:
the link is wrong

adrianha, 2018-01-28:
Which one @sibelius ? Just followed link on this: https://github.com/react-native-community/react-native-camera#how-to-use-master-branch
I already tried to replace with "git+https://github.com/react-native-community/react-native-camera.git" also not working

jgcmarins, 2018-01-28:
@adrianha after pointing to GitHub, you must remove all node_modules and install all over again.

adrianha, 2018-01-29:
Just tried that @jgcmarins , its still not working though

jqn, 2018-02-15:
For those still struggling with this please follow the [migration guid](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md) and remove old instances of RCTCamera

jqn, 2018-02-17:
is anyone else getting the annoying yellow warning after removing the FaceDetector?

ErickVoodoo, 2018-02-17:
@jqn Yes, I think it will be hidden in a Production mode. Don't be worried)

sibelius, 2018-02-17:
which yellow warning?

ErickVoodoo, 2018-02-17:
[image]
This one

sibelius, 2018-02-17:
@jgfidelis should we remove this warning?

maybe only loggind to ios log instead

jqn, 2018-02-17:
I vote yes, please remove this warning. It's already a fact that I removed the FaceDetector.

ajaxmac, 2018-02-22:
To get rid of the yellow warning, add to `index.ios.js`:
`console.disableYellowBox = true;`

jqn, 2018-02-22:
@ajaxmac yellow warning a still needed when developing, it is the kind of yellow warnings that don't make sense the ones that don't need to be there. I already know the facedetector is not integrated because I removed it myself.