react-native-camera/react-native-camera #1111: [0.13.0] Cannot build Android app - Manifest merger failed

joncursi, 2018-01-17:
My project builds fine on `0.12.0` but when I upgrade to `0.13.0`, I get this error while building the Android app:

```
:app:mergeDebugResources
:app:bundleDebugJsAndAssets SKIPPED
:app:copyDebugIconFonts UP-TO-DATE
:app:createDebugCompatibleScreenManifests UP-TO-DATE
:app:processDebugManifest
/Users/joncursi/Sites/cheddur/native/android/app/src/main/AndroidManifest.xml:25:13-35 Error:
	Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at AndroidManifest.xml:23:9-25:38 to override.

See http://g.co/androidstudio/manifest-merger for more information about the manifest merger.

:app:processDebugManifest FAILED
:app:buildInfoGeneratorDebug

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
  	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
  	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at AndroidManifest.xml:23:9-25:38 to override.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 6s

```

Any ideas on how to fix this? It's saying to add `'tools:replace="android:value"` somewhere, but I'm not sure exactly where. Thanks!

sibelius, 2018-01-17:
follow this https://github.com/react-native-community/react-native-camera#meta-data-android-26

closed 2018-01-17 by sibelius

malie, 2018-02-20:
that's on v0.13, no? https://github.com/react-native-community/react-native-camera/tree/v0.13.0#meta-data-android-26

harrydema, 2018-04-09:
Link fixed: 
[https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md#meta-data-android-26](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md#meta-data-android-26)

MaheshNandam, 2018-04-11:
Adding following lines to the build file works for me,

implementation (project(':react-native-camera')) {
    exclude group: "com.google.android.gms"
}
//noinspection GradleCompatible
implementation "com.google.zxing:core:3.2.1"
implementation "com.drewnoakes:metadata-extractor:2.9.1"
//noinspection GradleCompatible
implementation 'com.google.android.gms:play-services-vision:12.0.1'
implementation 'com.github.react-native-community:cameraview:cc47bb28ed2fc54a8c56a4ce9ce53edd1f0af3a5'