react-native-camera/react-native-camera #1112: ios crash when scan barcode many times

picksomething, 2018-01-18:
### Steps to reproduce
1.open camera to scan barcode
2.close camera after finish scanning
3.reopen camera to scan barcode

### Expected behaviour
Work as well

### Actual behaviour
Crash

### Environment
- **Node.js version**:v6.11.3
- **React Native version**:0.48.4
- **React Native platform + platform version**: iOS 11.1.2, iPhone x, iPhone 6s

### react-native-camera
**Version**:  "master"

### crash log
```
Incident Identifier: 77357851-2BE1-4DF9-BF62-E88444270145
Beta Identifier:     5AA92AC4-B9CB-4283-A406-77661F1240F5
Hardware Model:      iPhone10,3
Process:             express_cross_platform [4289]
Path:                /private/var/containers/Bundle/Application/E35E6BF3-CB95-4942-992B-1362E55949F6/express_cross_platform.app/express_cross_platform
Identifier:          com.eacent.kuaidihui
Version:             42 (2.2.1)
Beta:                YES
Code Type:           ARM-64 (Native)
Role:                Foreground
Parent Process:      launchd [1]
Coalition:           com.eacent.kuaidihui [1323]

Date/Time:           2018-01-17 10:55:19.3798 +0800
Launch Time:         2018-01-17 10:39:33.0876 +0800
OS Version:          iPhone OS 11.1.2 (15B202)
Baseband Version:    1.41.00
Report Version:      104

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000
Exception Note:  EXC_CORPSE_NOTIFY
Triggered by Thread:  0

Last Exception Backtrace:
0   CoreFoundation                	0x181aa9d04 __exceptionPreprocess + 124 (NSException.m:166)
1   libobjc.A.dylib               	0x180cf8528 objc_exception_throw + 56 (objc-exception.mm:521)
2   CoreFoundation                	0x181aa9390 __NSFastEnumerationMutationHandler + 128 (NSEnumerator.m:133)
3   UIKit                         	0x18aedc2a4 -[UIView(Hierarchy) _makeSubtreePerformSelector:withObject:withObject:copySublayers:] + 372 (UIView.m:9727)
4   UIKit                         	0x18b1bff88 __UIViewWasRemovedFromSuperview + 136 (UIView.m:9232)
5   UIKit                         	0x18aedbad8 -[UIView(Hierarchy) removeFromSuperview] + 508 (UIView.m:9307)
6   express_cross_platform        	0x10229cd1c -[RCTCamera removeFromSuperview] + 96 (RCTCamera.m:107)
7   UIKit                         	0x18aedeab8 -[UIView dealloc] + 512 (UIView.m:3880)
8   express_cross_platform        	0x1023517dc -[RCTView dealloc] + 132
9   CoreFoundation                	0x181a31098 cow_cleanup + 112 (NSCollectionAux.h:71)
10  CoreFoundation                	0x181978370 -[__NSArrayM dealloc] + 68 (NSArrayM.m:462)
11  libobjc.A.dylib               	0x180d16138 (anonymous namespace)::AutoreleasePoolPage::pop(void*) + 836 (NSObject.mm:1656)
12  CoreFoundation                	0x18196fd30 _CFAutoreleasePoolPop + 28 (NSObject.m:1355)
13  CoreFoundation                	0x181a4fa94 __CFRunLoopRun + 2020 (CFRunLoop.c:3117)
14  CoreFoundation                	0x18196ffb8 CFRunLoopRunSpecific + 436 (CFRunLoop.c:3245)
15  GraphicsServices              	0x183807f84 GSEventRunModal + 100 (GSEvent.c:2245)
16  UIKit                         	0x18af442e8 UIApplicationMain + 208 (UIApplication.m:3949)
17  express_cross_platform        	0x1022a81e8 main + 92 (AppDelegate.swift:6)
18  libdyld.dylib                 	0x18149256c start + 4
```

FaiChou, 2018-10-15:
Same problem.

Even I tried to add judgment for `self.superview`:

[image]

I think it is in other queue which sets `superview` to nil,  so I tried to fix this by adding `main queue`:

```c
- (void)removeFromSuperview
{
    [self stopSession];
    dispatch_async(dispatch_get_main_queue(), ^{
      [super removeFromSuperview];
    });
    [[NSNotificationCenter defaultCenter] removeObserver:self name:UIDeviceOrientationDidChangeNotification object:nil];
}
```

FaiChou, 2018-10-16:
### Steps to reproduce

1. scan qrcode (or ean code)
2. leave scan page
3. repeat 1,2 many times

### Actual behaviour

Crash with native.

### RN core code

```javascript
  render() {
    if (this.state.isOk) {
      return (
        <View style={{
          flex: 1,
          backgroundColor: 'black',
        }} />
      )
    }
    return (
      <QRScannerView
        onScanResultReceived={(e) => this.barcodeReceived(e)}
        renderTopBarView={() => null}
        renderBottomMenuView={() => null}
      />
    );
  }
```

If `onScanResultReceived` gets result, then sets `isOk` state to `true`, this will cause `QRScannerView` unmount.

### Environment

```
Environment:
  OS: macOS 10.14
  Node: 9.5.0
  Yarn: 1.7.0
  npm: 5.7.1
  Watchman: 4.9.0
  Xcode: Xcode 10.0 Build version 10A255
  Android Studio: 3.1 AI-173.4720617

Packages: (wanted => installed)
  react: 16.3.1 => 16.3.1
  react-native: 0.55.4 => 0.55.4
```

FaiChou, 2018-10-21:
fixed with move `[self stopSession];` to bottom of function.

alimek, 2018-10-22:
anyone has fix for it ?

alimek, 2018-10-22:
@FaiChou are you able to give me a path how to reproduce this issue ?

It's happening for me, but not often. I would like to be able to reproduce this issue first and then check your suggestion, but sometimes i crashes for 1st time, and sometimes after 100x times scanning QR code.

sibelius, 2018-10-22:
@FaiChou can u send a PR fixing this?

alimek, 2018-10-22:
PR #1868 

but i am still checking if this works for me, but its hard for me to reproduce it everytime :(

FaiChou, 2018-10-23:
@sibelius  @alimek  

Check the new code of the RNCamera https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L133

I fixed it after checking the new code.

closed 2019-04-18 by sibelius