react-native-camera/react-native-camera #1113: Manifest tools:node="replace" cause FCM issue on android

thientnc-ibl, 2018-01-18:
In the README, we have the below instruction to fix Manifest error
```
<application
      android:name=".MainApplication"
      android:allowBackup="true"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:theme="@style/AppTheme"
      tools:node="replace"
    >
```
But this line `tools:node="replace"` causes the other critical bug, the android app cannot receive any push notification. The issue is here, it took me 2 hours for debugging.
https://stackoverflow.com/questions/39083566/fcm-returns-error-notregistered-after-the-first-message

### Expected behaviour
Please downgrade the version of `com.android.support:exifinterface` to `26.0.1`.

### Environment
- **React Native version**: 0.52.0
- **React Native platform**: Android

### react-native-camera
**Version**: "master"

jbarradas, 2018-02-06:
I'm having the same issue! Using react-native-onesginal

Any news on this? How did you solve this issue?

Thanks!

jgfidelis, 2018-02-06:
Can you guys try making your app/build.gradle look like this:
```
    compile (project(':react-native-camera')) {
        exclude group: "com.android.support"
    }
```

Can you try compiling?

jbarradas, 2018-02-06:
@jgfidelis Yes, it worked for me. Although I had to put it at the bottom of the dependencies list. Otherwise it didn't compile. 

Cheers! ;)

thientnc-ibl, 2018-02-07:
@jbarradas Thank for your comment `Although I had to put it at the bottom of the dependencies list`. You save my life

francescortiz, 2018-02-14:
The Merged Mafinest got totally screwed up because of this bad advice. **tools:node="replace"** should be removed from your android manifest. One week lost because of this. At least. I could bypass some other issues, but with FCM it was a pain in the ass. Searching for `Other Manifest Files (Included in merge, but did not contribute any elements)` did not help at all, so I write this comment just in case someone else falls here from a Google search.

patrickalbani, 2018-04-23:
I resolve removing tools:node="replace" and adding in build.gradle:

```
 configurations.all {
    resolutionStrategy.eachDependency { DependencyResolveDetails details ->
        def requested = details.requested
        if (requested.group == 'com.android.support') {
            if (!requested.name.startsWith("multidex")) {
                details.useVersion '26.0.0'
            }
        }
    }
```

nikitamarysoloman, 2018-06-12:
try to replace in your code

with the below attributes
android:allowBackup="false"
tools:node="merge"
tools:replace="android:label,android:allowBackup"

tcjuan, 2018-07-08:
After I spent two days wandering around and pouch keyboard day and night. The above solutions were all not working.
I finally try to digging the code and find out it is because firebase ContentProvider loading mechanism issue.
I find out it is beacuse I have on "contentprovider" need to load that prevent firebase content provider loading. 
their is a solution can prevent firebase ContentProvider auto load.
Just put in app/AndroidManifest.xml
<provider
    android:name="com.google.firebase.provider.FirebaseInitProvider"
    android:authorities="${applicationId}.firebaseinitprovider"
    tools:node="remove"
    />
In your app @oncreate, manual init FirebaseAPP.
It is working!

closed 2019-06-08 by stale[bot]