react-native-camera/react-native-camera #1115: Camera view freezes after taking one photo

Tsiniloiv, 2018-01-18:
### Steps to reproduce

Camera render code (Excuse the terribly messy layout. I am not great at working with layout things):

```
renderCamera() {
    if(!this.state.showGallery) {
      return (
            <Camera ref={ref => {
              this.camera = ref;
            }} style={styles.overlaycontainer}
            captureTarget={Camera.constants.CaptureTarget.cameraRoll}
            aspect={Camera.constants.Aspect.fit}
            orientation={Camera.constants.Orientation.auto}
            defaultOnFocusComponent={true}
            >
                <View style={{width: 225}}></View>
                <View style={[styles.guidebar]}></View>
                <View style={{width: 225, flexDirection: 'column', justifyContent: 'space-between'}}>
                  <View style={{flexDirection: 'row', justifyContent: 'space-between'}}>
                    <View></View>
                    <View style={{flexDirection: 'column'}}>
                      <Text style={[styles.capture]} onPress={this.takePicture.bind(this)}>[PHOTO]</Text>
                      {(!this.state.isRecording && (
                        <Text style={[styles.capture]} onPress={this.startRecording.bind(this)}>[RECORD]</Text>
                      )) || (
                        <Text style={[styles.capture]} onPress={this.stopRecording.bind(this)}>[!RECORD]</Text>
                      )}
                    </View>
                  </View>
                  <View style={{flexDirection: 'row', justifyContent: 'space-between'}}>
                    <View></View>
                    <Text style={[styles.capture]} onPress={this.toggleView.bind(this)}>[GALLERY]</Text>
                  </View>
                </View>
            </Camera>
        );
    }
  }
```

Capture code:

```
takePicture = async function() {
    const options = {};
    //options.location =  ...
    if(this.camera) {
      this.camera.capture({metadata: options})
        .then((data) => console.log(data))
        .catch(err => console.log(err));
        //console.log('Picture taken!');
    }
  };

  startRecording = () => {
    if (this.camera) {
      this.camera
        .capture({ mode: Camera.constants.CaptureMode.video })
        .then(data => console.log(data))
        .catch(err => console.error(err));
      this.setState({
        isRecording: true,
      });
      console.log("Recording!");
    }
  };

  stopRecording = () => {
    if (this.camera) {
      this.camera.stopCapture();
      this.setState({
        isRecording: false,
      });
      console.log("No longer recording!");
    }
  };
```

### Expected behaviour
Pressing the [PHOTO] button should take a picture, saving that picture to the DCIM folder. I should then be able to do it again, and again, and again, etc.

### Actual behaviour
I press the [PHOTO] button once, and it runs as expected. The console outputs a uri and path, the picture is saved to my DCIM folder. However, the camera view freezes completely when the picture is taken, and does not resume.

Pressing the [PHOTO] button again plays the camera shutter sound, but does not save an image. Attempting to record a video when the camera view is frozen throws an error. Attempting to record a video when the camera view is not frozen works as intended.

Turning takePicture into an arrow function like startRecording and stopRecording are does nothing.

### Environment
- **Node.js version**: 9.3.0
- **React Native version**: 0.52.0
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**: ^0.13.0

kyle-ssg, 2018-01-18:
+1 can replicate after upgrading from 0.12 to 0.13

sibelius, 2018-01-25:
can u try RNCamera?

asaenvoy, 2018-02-23:
Switching from
```JS
import Camera from 'react-native-camera';
```
to:
```JS
import { RNCamera } from 'react-native-camera';
```
fixed this issue for me.

closed 2018-02-23 by sibelius

andrejunqueira1, 2018-07-10:
My workaround was
  call `this.camera.stopPreview()` after call `this.camera.startPreview()`