react-native-camera/react-native-camera #1120: Changes needed for gradle 3.0.1 compatibility

pacamara, 2018-01-19:
### Steps to reproduce
1. Build downstream project https://github.com/status-im/status-react
2. Observe this build warning:
```
Configuration 'compile' in project ':react-native-camera' is deprecated. Use 'implementation' instead.
```

### Expected behaviour
react-native-camera uses gradle 3.0.1 compliant syntax and there is no build warning.
Since `api` is recommended for library projects, suggest using that, rather than `implementation`.

### Actual behaviour
The build warning in step (2) above is observed.

### Environment
- **Node.js version**: 9.4.0
- **React Native version**:  0.51.0
- **React Native platform + platform version**: Ubuntu 16.04 LTS, Android build
### react-native-camera
**Version**: 0.10.0 (problem still exists in 0.13.0)

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sibelius, 2018-01-19:
Can u send a PR?

pacamara, 2018-01-25:
@sibelius Hi, apologies for the slow reply, and thanks @jgfidelis  for doing PR #114, which I see is now rolled back.

We now realize this is a breaking change for new projects created with `react-native init`, unless manual changes are made to the generated project (see https://github.com/react-native-community/react-native-svg/issues/581#issuecomment-359201096 ). Probably why #114 was rolled back? We'll wait for react-native to upgrade to gradle 3.

sibelius, 2018-01-25:
we can reopen when react native is using gradle 3

closed 2018-01-25 by sibelius