react-native-camera/react-native-camera #1122: Error when archiving on MASTER: /Users/scoob/odi-react-native/node_modules/react-native-camera/ios/RCTCameraManager.m:706:54: Use of undeclared identifier 'viewportSize'

ryd0rz, 2018-01-19:
### Steps to reproduce
1. use master latest
2. archive ios project

### Expected behaviour
shoud not throw error

### Actual behaviour
throws error. please test your pull request before merge.
Here is my PR comment (however, the PR is already merged, and needs fixed):
https://github.com/react-native-community/react-native-camera/commit/0c4161ab1749e236f9773a6ef5ba95dbc0c519e7#diff-67aecadd1220a0eebd27fd0e61cd2b91R706

I had to use master cause of other issues related to not working in latest version of RN.

### Environment
"react": "16.0.0-alpha.12",
"react-native": "0.51.0",
"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git",

### react-native-camera
**Version**: master

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

ryd0rz, 2018-01-19:
PR: https://github.com/react-native-community/react-native-camera/pull/1123

jgcmarins, 2018-01-20:
thanks @ryd0rz

closed 2018-01-20 by jgcmarins