react-native-camera/react-native-camera #1125: Upgrade to 0.13.0 throws `undefined is not an object` for CameraManager.FaceDetection

gmeluski, 2018-01-20:
### Steps to reproduce
1. `yarn add react-native-camera@0.13.0`
2. `react-native unlink react-native-camera && react-native link react-native-camera` (this is coming from 0.12.0, I have also attempted manual linking)
3. Build project on simulator - even try cleaning xcode project, stopping the react server and doing a `watchman watch-del-all` before restarting the react server and building the xcode project.

### Expected behaviour
App should run - it was working when using 0.12.0. 

### Actual behaviour
Red screen:
```
undefined is not an object (evaluating 'CameraManager.FaceDetection.Mode')

<unknown>
    undefined:273:51
loadModuleImplementation
    require.js:213:12
guardedLoadModule
    require.js:140:45
accept
    require.js:333:12
define
    require.js:78:24
global code
    undefined:1:4
<unknown>
    HMRClient.js:112:27
onmessage
    HMRClient.js:102:26
dispatchEvent
    event-target.js:172:43
<unknown>
    WebSocket.js:237:27
emit
    EventEmitter.js:182:12
__callFunction
    MessageQueue.js:353:47
<unknown>
    MessageQueue.js:118:26
__guardSafe
    MessageQueue.js:316:6
callFunctionReturnFlushedQueue
    MessageQueue.js:117:17
```
I have also tried removing a few offending lines from the code, however it also gives `undefined is not an object` for `CameraManager.Type.back`

### Environment
- **Node.js version**: 9.3.0
- **React Native version**: 0.52.0-rc.0 (also tried with 0.52.0)
- **React Native platform + platform version**: iOS 11.2

### react-native-camera
**Version**: 0.13.0

thientnc-ibl, 2018-01-23:
I got this error too. Actually, the `CameraManager` and `CameraModule` are `undefined` when getting from NativeModules
in Camera.js file
```
const CameraManager = NativeModules.CameraManager || NativeModules.CameraModule
```

sibelius, 2018-01-23:
Fixed on master

closed 2018-01-23 by sibelius