react-native-camera/react-native-camera #1129: Execution failed for task ':react-native-camera:processDebugAndroidTestManifest'.

evansherry, 2018-01-21:
### Steps to reproduce
1. Start marke project , but build fail
2. Follow Readme , add tools:node="replace" to AndroidManifest.xml, add this to build.gradle(app)
```
 compile (project(':react-native-camera')) {
        exclude group: "com.android.support"
}
 compile ('com.android.support:exifinterface:26.0.1') {
        force = true;
    }

configurations.all {
    resolutionStrategy.eachDependency { DependencyResolveDetails details ->
        def requested = details.requested
        if (requested.group == 'com.android.support') {
            details.useVersion "${rootProject.supportLibraryVersion}"
        }
    }
}
```
3. sdk versions:
    compileSdkVersion = 26
    minSdkVersion = 16
    targetSdkVersion = 26
    supportLibraryVersion = '26.1.0'

### Expected behaviour
build successful

### Actual behaviour
FAILURE: Build failed with an exception.

* What went wrong:
build fail !
Execution failed for task ':react-native-camera:processDebugAndroidTestManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
  	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
  	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at manifestMerger2304450501655821264.xml:23:9-25:38 to override.

### Environment
-** OS**: Mac OS Sieera
- **Node.js version**:   v7.7.4
- **React Native version**:  0.46.1
- **React Native platform + platform version**:  Android 6.0.1 

### react-native-camera
**Version**: npm version  4.1.2
When I set up add  'tools:replace="android:value" to AndroidManifest.xml, build fail, remove continue 'tools:replace="android:value" to build success, but I do not want to always be the case

Samykills, 2018-01-22:
Stuck with same issue? :(
Do u have any other dependencies in your project...i have react-native-google-places which i suspect might be causing some problems!

Samykills, 2018-01-22:
build.gradle for react-native-camera has 
dependencies {
  ...
  compile "com.android.support:exifinterface:26.0.2"
}

this is leads to ..
Error:Execution failed for task ':react-native-camera:processDebugAndroidTestManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
  	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
  	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at manifestMerger4058125526788706578.xml:23:9-25:38 to override.

evansherry, 2018-01-23:
Sorry, I'm late, and finally I did not use studio run, I'm in WebStorm run android and ./gradlew + other commands.

sibelius, 2018-01-23:
follow this https://github.com/react-native-community/react-native-camera#meta-data-android-26

closed 2018-01-23 by sibelius

ThomLaGreca, 2018-01-31:
I followed the above steps and still i get the same error. Every post i find regarding this issue just points to the above meta-data-android-26 link and stops... Help. 

> Error:Execution failed for task ':react-native-camera:processDebugAndroidTestManifest'.
> > Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
>   	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
>   	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at manifestMerger1679199706524611983.xml:23:9-25:38 to override.

```
dependencies {
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:26.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
    compile (project(':react-native-camera')) {
        exclude group: "com.android.support"

        // uncomment this if also com.google.android.gms:play-services-vision versions are conflicting
        // this can happen if you use react-native-firebase
        // exclude group: "com.google.android.gms"
    }

    compile ('com.android.support:exifinterface:26.0.1') {
        force = true;
    }

    // uncomment this if you uncommented the previous line
    // compile ('com.google.android.gms:play-services-vision:11.6.0') {
    //    force = true;
    // }
}
```

thevishnupradeep, 2018-02-04:
@ThomLaGreca meta-data-android-26 link actually helps. add
`exclude group: "com.google.android.gms"`
and add
`maven {
     url "https://jitpack.io"
}`
to your project level build.gradle

pielambr, 2018-02-11:
@ThomLaGreca Did you ever fix this in the end? I have the same problem as you have

Taym95, 2018-02-12:
Same issue here!

prjmax, 2018-02-17:
Same problem, I'm not sure if this is related, I can run the App in Dev mode but when I try to execute assembleRelease I get a "Failed to execute aapt" error

sibelius, 2018-02-18:
You should disable appt now

In the next React native version 54, it will solve the issue with appt2

prjmax, 2018-02-18:
you are right @sibelius, adding android.enableAapt2=false in gradle.properties solved the issue, Thanks!

sibelius, 2018-02-18:
@prjmax could you send a PR to https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md

adding this note about `aapt2`?

afilp, 2018-02-21:
@sibelius Whatever I do I always have the problem that all others mentioned. The Q&A solutions do not fix the problem. Can you please let us know what else to do?

h-sakano, 2018-02-26:
In my case, when I change the version specification of "com.android.support:exifinterface", the build will now succeed.

```
dependencies {
    ...
    // compile "com.android.support:exifinterface:26.0.2"
    compile "com.android.support:exifinterface:26.0.1"
}
```

satish754ss, 2018-03-02:
Error:Execution failed for task ':react-native-camera:processDebugAndroidTestManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35
	is also present at [com.android.support:support-v4:26.0.1] AndroidManifest.xml:28:13-35 value=(26.0.1).
	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at manifestMerger1412991257447853314.xml:23:9-25:38 to override. 
please help to solve this problem

arapocket, 2018-03-05:
Adding `android.enableAapt2=false` didn't fix anything for me.

cvvspb, 2018-03-06:
Looks like following the Q/A section https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md **and updating from 1.0.0 to 1.0.1** fixed the issue for me.

GarryLaly, 2018-04-28:
Following this step https://medium.com/react-native-zone/error-react-native-camera-com-android-support-exifinterface-26-0-2-7e779366c828 I tested is work.