react-native-camera/react-native-camera #1131: BarCodeReadEvent.java:27: error: no suitable method found for init(int)

sasha-stojanovic, 2018-01-21:
### Steps to reproduce
1. react-native init newProject
2. cd newProject
3. npm install react-native-camera --save
4. npm link react-native-camera
5. react-native run-android

### Expected behaviour
The app should compile onto the android emulator 

### Actual behaviour
It crashes at 32% build with **"BarCodeReadEvent.java:27: error: no suitable method found for init(int)"** .... I can run and use the camera functionality perfectly when deploying to IOS.

### react-native-camera
I have changed to master now to no avail.

NOTE - I have done all items such as add permissions and add the excludes to the gradle files as per your readme.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgcmarins, 2018-01-21:
@sasha-stojanovic which `react-native-camera` version?

sasha-stojanovic, 2018-01-21:
@jgcmarins see below all my dependencies and their versions from my package.json

"dependencies": {
    "react": "16.2.0",
    "react-native": "0.52.0",
    "react-native-action-button": "^2.8.3",
    "react-native-camera": "^0.13.0",
    "react-native-elements": "^0.19.0",
    "react-native-vector-icons": "^4.5.0",
    "react-navigation": "^1.0.0-beta.27"
  },

sasha-stojanovic, 2018-01-22:
Any ideas? Could it be something with my android sdk or is it react-native-camera related?

Has anyone seen this error before?

jgcmarins, 2018-01-22:
we've made some changes, could you try to run on `master branch`, please?
https://github.com/react-native-community/react-native-camera#how-to-use-master-branch

sasha-stojanovic, 2018-01-23:
@jgcmarins I have switched to master branch and had to change the dependencies in your android/build.gradle to compile.

I am still getting same error now when compiling to android. 
[image]

sibelius, 2018-01-23:
this happens because android is getting react-native from gradle instead of node_modules

add this to your gradle config

```
maven {
    // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
    url "../node_modules/react-native/android"
  }
```

sasha-stojanovic, 2018-01-23:
I used to have jitpack in the line above. I have changed it to node_modules and now I am getting  ... 

> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
> Could not find com.github.react-native-community:cameraview:df60b07573.

luqingxuan, 2018-02-09:
@sasha-stojanovic 
i have the same error,  did u find answer for this problem ?

JoshiiSinfield, 2018-02-09:
@sibelius @sasha-stojanovic I'm also seeing the issue, as described above.

react-native still references android sdk 23. react-native-camera sets dependencies for 26.0.1.

I tried using maven.google.com as maven url in gradle in order to resolve not finding exifinterface which you reference as 26.0.2. That fixed the missing exifinterface library, but leaves the cameraview issue. so I added jitpack url, and instead get the same java compilation errors as mentioned above. 

This is the grade dependency issues I got:

A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:exifinterface:26.0.2.
     Searched in the following locations:
     Required by:
         MyCamera:app:unspecified > MyCamera:react-native-camera:unspecified
   > Could not find com.github.react-native-community:cameraview:df60b07573.
     Required by:
         MyCamera:app:unspecified > MyCamera:react-native-camera:unspecified

This is my javac errors:

:react-native-camera:compileReleaseJavaWithJavac
:react-native-camera:compileReleaseJavaWithJavac - is not incremental (e.g. outputs have changed, no previous execution, etc.).
/Users/joshs/git/MyCamera/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarCodeReadEvent.java:27: error: no suitable method found for init(int)
    event.init(viewTag);
         ^
    method Event.init(int,long) is not applicable
      (actual and formal argument lists differ in length)
    method BarCodeReadEvent.init(int,Result) is not applicable
      (actual and formal argument lists differ in length)
/Users/joshs/git/MyCamera/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/BarCodeReadEvent.java:32: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/joshs/git/MyCamera/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/CameraReadyEvent.java:22: error: method init in class Event<T> cannot be applied to given types;
    event.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/joshs/git/MyCamera/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/FaceDetectionErrorEvent.java:24: error: no suitable method found for init(int)
    event.init(viewTag);
         ^
    method Event.init(int,long) is not applicable
      (actual and formal argument lists differ in length)
    method FaceDetectionErrorEvent.init(int,RNFaceDetector) is not applicable
      (actual and formal argument lists differ in length)
/Users/joshs/git/MyCamera/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/FaceDetectionErrorEvent.java:29: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/joshs/git/MyCamera/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/CameraMountErrorEvent.java:22: error: method init in class Event<T> cannot be applied to given types;
    event.init(viewTag);
         ^
  required: int,long
  found: int
  reason: actual and formal argument lists differ in length
  where T is a type-variable:
    T extends Event declared in class Event
/Users/joshs/git/MyCamera/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/FacesDetectedEvent.java:52: error: method init in class Event<T> cannot be applied to given types;
    super.init(viewTag);
         ^

JoshiiSinfield, 2018-02-09:
Hmm. I've just ran 
https://github.com/react-native-community/rncamera-example successfully. 

Seems that dependencies & versions may be different?

sibelius, 2018-02-09:
fixed on master

closed 2018-02-09 by sibelius

luqingxuan, 2018-02-11:
@sasha-stojanovic 
after i add this, it seems work, maybe it will help u

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven { url 'https://jitpack.io' }
        maven {
            url "https://maven.google.com"
        }
    }
}