react-native-camera/react-native-camera #1133: RNCameraExample   Application startup crash

fadeli1, 2018-01-22:
### Steps to reproduce
1. run RNCameraExample
2.Application startup crash
3.

### Expected behaviour
 run RNCameraExample success

### Actual behaviour
Application startup crash

### Environment
- **Node.js version**:v6.3.0
- **React Native version**:0.51.0
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**:  0.13.0

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

fadeli1, 2018-01-22:
my file build.gradle:
apply plugin: "com.android.application"

import com.android.build.OutputFile

/**
 * The react.gradle file registers a task for each build variant (e.g. bundleDebugJsAndAssets
 * and bundleReleaseJsAndAssets).
 * These basically call `react-native bundle` with the correct arguments during the Android build
 * cycle. By default, bundleDebugJsAndAssets is skipped, as in debug/dev mode we prefer to load the
 * bundle directly from the development server. Below you can see all the possible configurations
 * and their defaults. If you decide to add a configuration block, make sure to add it before the
 * `apply from: "../../node_modules/react-native/react.gradle"` line.
 *
 * project.ext.react = [
 *   // the name of the generated asset file containing your JS bundle
 *   bundleAssetName: "index.android.bundle",
 *
 *   // the entry file for bundle generation
 *   entryFile: "index.android.js",
 *
 *   // whether to bundle JS and assets in debug mode
 *   bundleInDebug: false,
 *
 *   // whether to bundle JS and assets in release mode
 *   bundleInRelease: true,
 *
 *   // whether to bundle JS and assets in another build variant (if configured).
 *   // See http://tools.android.com/tech-docs/new-build-system/user-guide#TOC-Build-Variants
 *   // The configuration property can be in the following formats
 *   //         'bundleIn${productFlavor}${buildType}'
 *   //         'bundleIn${buildType}'
 *   // bundleInFreeDebug: true,
 *   // bundleInPaidRelease: true,
 *   // bundleInBeta: true,
 *
 *   // whether to disable dev mode in custom build variants (by default only disabled in release)
 *   // for example: to disable dev mode in the staging build type (if configured)
 *   devDisabledInStaging: true,
 *   // The configuration property can be in the following formats
 *   //         'devDisabledIn${productFlavor}${buildType}'
 *   //         'devDisabledIn${buildType}'
 *
 *   // the root of your project, i.e. where "package.json" lives
 *   root: "../../",
 *
 *   // where to put the JS bundle asset in debug mode
 *   jsBundleDirDebug: "$buildDir/intermediates/assets/debug",
 *
 *   // where to put the JS bundle asset in release mode
 *   jsBundleDirRelease: "$buildDir/intermediates/assets/release",
 *
 *   // where to put drawable resources / React Native assets, e.g. the ones you use via
 *   // require('./image.png')), in debug mode
 *   resourcesDirDebug: "$buildDir/intermediates/res/merged/debug",
 *
 *   // where to put drawable resources / React Native assets, e.g. the ones you use via
 *   // require('./image.png')), in release mode
 *   resourcesDirRelease: "$buildDir/intermediates/res/merged/release",
 *
 *   // by default the gradle tasks are skipped if none of the JS files or assets change; this means
 *   // that we don't look at files in android/ or ios/ to determine whether the tasks are up to
 *   // date; if you have any other folders that you want to ignore for performance reasons (gradle
 *   // indexes the entire tree), add them here. Alternatively, if you have JS files in android/
 *   // for example, you might want to remove it from here.
 *   inputExcludes: ["android/**", "ios/**"],
 *
 *   // override which node gets called and with what additional arguments
 *   nodeExecutableAndArgs: ["node"],
 *
[... 72 lines cut ...]

           // uncomment this if also com.google.android.gms:play-services-vision versions are conflicting
           // this can happen if you use react-native-firebase
           // exclude group: "com.google.android.gms"
       }

       compile ('com.android.support:exifinterface:26.0.1') {
           force = true;
       }

       // uncomment this if you uncommented the previous line
       // compile ('com.google.android.gms:play-services-vision:11.6.0') {
       //    force = true;
       // }
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile "com.facebook.react:react-native:+"  // From node_modules
}

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
task copyDownloadableDepsToLibs(type: Copy) {
    from configurations.compile
    into 'libs'
}

fadeli1, 2018-01-22:
my file AndroidManifest.xml:
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
          xmlns:tools="http://schemas.android.com/tools"
    package="com.rncameraexample"
    android:versionCode="1"
    android:versionName="1.0">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

    <uses-sdk
        android:minSdkVersion="16"
        android:targetSdkVersion="26" />

    <application
      android:name=".MainApplication"
      android:allowBackup="true"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:theme="@style/AppTheme"
      tools:node="replace"
    >
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenSize"
        android:windowSoftInputMode="adjustResize">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
      <activity android:name="com.facebook.react.devsupport.DevSettingsActivity" />
    </application>

</manifest>

fadeli1, 2018-01-22:
studio caught the error message:
01-22 18:22:53.643 22111-22162/com.rncameraexample W/System: ClassLoader referenced unknown path: /system/app/webview/lib/arm
01-22 18:22:53.979 22111-22111/com.rncameraexample W/ResourceType: No known package when getting name for resource number 0xffffffff
01-22 18:22:53.979 22111-22111/com.rncameraexample W/ResourceType: No known package when getting name for resource number 0xffffffff
01-22 18:22:54.046 22111-22111/com.rncameraexample E/AndroidRuntime: FATAL EXCEPTION: main
                                                                     Process: com.rncameraexample, PID: 22111
                                                                     java.lang.NoSuchMethodError: No static method getDisplay(Landroid/view/View;)Landroid/view/Display; in class Landroid/support/v4/view/ViewCompat; or its super classes (declaration of 'android.support.v4.view.ViewCompat' appears in /data/app/com.rncameraexample-1/base.apk)
                                                                         at com.google.android.cameraview.CameraView.onAttachedToWindow(CameraView.java:135)
                                                                         at android.view.View.dispatchAttachedToWindow(View.java:14937)
                                                                         at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2962)
                                                                         at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2969)
                                                                         at android.view.ViewGroup.addViewInner(ViewGroup.java:4477)
                                                                         at android.view.ViewGroup.addView(ViewGroup.java:4274)
                                                                         at android.view.ViewGroup.addView(ViewGroup.java:4215)
                                                                         at com.facebook.react.uimanager.ViewGroupManager.addView(ViewGroupManager.java:47)
                                                                         at com.facebook.react.uimanager.NativeViewHierarchyManager.manageChildren(NativeViewHierarchyManager.java:401)
                                                                         at com.facebook.react.uimanager.UIViewOperationQueue$ManageChildrenOperation.execute(UIViewOperationQueue.java:180)
                                                                         at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:819)
                                                                         at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:926)
                                                                         at com.facebook.react.uimanager.UIViewOperationQueue.access$2100(UIViewOperationQueue.java:47)
                                                                         at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:986)
                                                                         at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
                                                                         at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:136)
                                                                         at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
                                                                         at android.view.Choreographer$CallbackRecord.run(Choreographer.java:897)
                                                                         at android.view.Choreographer.doCallbacks(Choreographer.java:701)
                                                                         at android.view.Choreographer.doFrame(Choreographer.java:633)
                                                                         at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:885)
                                                                         at android.os.Handler.handleCallback(Handler.java:815)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:104)
                                                                         at android.os.Looper.loop(Looper.java:207)
                                                                         at android.app.ActivityThread.main(ActivityThread.java:5975)
                                                                         at java.lang.reflect.Method.invoke(Native Method)
                                                                         at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888)
                                                                         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:749)
01-22 18:22:54.048 1104-2493/? W/ActivityManager:   Force finishing activity com.rncameraexample/.MainActivity
01-22 18:22:54.582 1104-1120/? W/ActivityManager: Activity pause timeout for ActivityRecord{7022672 u0 com.rncameraexample/.MainActivity, isShadow:false t90 f}

closed 2018-01-22 by fadeli1

404sand808s, 2018-02-20:
Hey @fadeli1 , could you elaborate on how this fixed your issue? I have the same error it seems occurring when I snap a photo on Android, but adding that to AndroidManifest didn't seem to make any difference for me.

arapocket, 2018-03-15:
I have the same error as @fadeli1 but his solution won't work for me because I already have a meta-data tag with different values.

@404sand808s I'm not sure we can have multiple meta-data tags