react-native-camera/react-native-camera #1139: Build error on upgrading to version 0.13.0

khrizt, 2018-01-22:
### Steps to reproduce
1. Upgrade to react-native-camera 0.13.0
2. Implement meta-data android 26 fix
3. Build android app

### Expected behaviour
Build successfully

### Actual behaviour
* What went wrong:
Execution failed for task ':app:transformClassesWithJarMergingForDebug'.
> com.android.build.api.transform.TransformException: java.util.zip.ZipException: duplicate entry: com/google/firebase/FirebaseApiNotAvailableException.class

### Environment
- **Node.js version**: 9.2.0
- **React Native version**: 0.52.0
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**: 0.13.0

I'm also using `firebase-core` and `firebase-analytics`, I don't know of there's something more to exclude on the dependencies, but I couldn't figure it out.

sibelius, 2018-01-22:
check this https://github.com/react-native-community/react-native-camera#as-the-error-message-hints-comandroidsupportexifinterface2602-is-already-found-in-comandroidsupportsupport-v42601

khrizt, 2018-01-22:
Thanks for replying, sorry I forgot to add this step, I already did that. The only difference is that I use support library v7, but I don't think this should make much difference.

closed 2018-12-29 by n1ru4l