react-native-camera/react-native-camera #1149: camera.capture is not return base64 string?

pacozaa, 2018-01-23:
Here is my `takePicture` function as in the example. 
``` javascript
takePicture() {
    const options = {};
    //options.location = ...
    this.camera.capture({ metadata: options })
      .then((data) => {
        console.log(data)
      })
      .catch(err => console.error(err))    
  }
```

`console.log ` inform me that `data ` is 
```
Object {mediaUri: "content://media/external/images/media/14071", path: "file:///storage/emulated/0/DCIM/IMG_20180123_212416.jpg"}
```

What am I missing? I want to get based64 string to cooperate with `<Image/>`

sibelius, 2018-01-23:
rctcamera or rncamera?

pacozaa, 2018-01-23:
@sibelius RCTCameraPackage

pacozaa, 2018-01-23:
@sibelius wait, What do you mean by rctcamera or rncamera?

pacozaa, 2018-01-24:
Yes, It is `RCTCamera` in `MainApplication.java`
`import com.lwansbrough.RCTCamera.RCTCameraPackage;`

pacozaa, 2018-01-24:
OK, I solve this problem myself. According to #278 #892 , I tried and tested their answers and come up with something different. 

Here is what I have learnt.

1.  I don't need `base64` to show picture in `<Image>`
2. `data.path` can used in `uri`
3. Don't bother convert `data` from `this.camera.capture`if you just want to show picture in your `<Image>`

Basically, I ask wrong question.

For example,

``` javascript
takePicture = () => {
    const options = {}
    //options.location = ...
    this.camera.capture({ metadata: options })
      .then((data) => {
        this.setState({
           data: data
        })
      })
      .catch(err => console.error(err))
  }

<Image
 source={{ uri: this.state.data.path }}
/> 
```

Cheers!

closed 2018-01-24 by pacozaa