react-native-camera/react-native-camera #1151: ios build fail 'GoogleMVDataOutput/GoogleMVDataOutput.h' file not found

evansherry, 2018-01-23:
RCTCamera or RNCamera?
RCTCamera

### Steps to reproduce
1. build in xcode ,fail msg 'GoogleMVDataOutput/GoogleMVDataOutput.h' file not found

### Expected behaviour
Tell us what should happen
build successful

### Actual behaviour
Tell us what happens instead
'GoogleMVDataOutput/GoogleMVDataOutput.h' file not found

### Environment
- **OS**: Mac OS Sieera
- **Node.js version**:   v7.7.4
- **React Native version**:  0.46.1
- **React Native platform + platform version**: ...

### react-native-camera
**Version**: npm version  4.1.2

[image]

sibelius, 2018-01-23:
you need to use podfile on ios using master

check this https://github.com/react-native-community/react-native-camera/blob/master/RNCameraExample/ios/Podfile#L1

or try to link manually

jgfidelis, 2018-01-23:
Create a Podfile inside your ios folder. The content should be:
target “TARGET_NAME” do
  pod 'GoogleMobileVision/FaceDetector'
  pod 'GoogleMobileVision/MVDataOutput'
end

in the example, TARGET_NAME is "RNCameraExample".

Run pod install. Open the workspace and run your target

jgfidelis, 2018-01-23:
After PR #1153 , try to follow the GMV installation steps in the readme and let us know if it fixes for you.

closed 2018-01-23 by sibelius

evansherry, 2018-01-24:
@sibelius podfile not found ,Are you sure you push it?

evansherry, 2018-01-24:
@jgfidelis  According to your tips, pod install is still not resolved
[image]
[image]

jgfidelis, 2018-01-24:
Hi, the podfile is not present anymore, we removed it. You now have to manually install the frameworks. This is a temporary thing, in the future we should make it optional, only people who should want to use FaceDetector should install the frameworks. For now, follow the steps on installing GMV on the readme.md. The RCTCamer project looks for headers and frameworks in the ios/Frameworks of your app project.

kesha-antonov, 2018-01-24:
Yes, please, make it optional @jgfidelis

kesha-antonov, 2018-01-24:
Instructions from readme doesn't work for me.

kesha-antonov, 2018-01-24:
Fixed by adding ```Framework Search Paths```
[image]

suijunqiang, 2018-01-26:
@jgfidelis
Thanks for you post.
I got same issue too, and I have no ideal how to add popfile & FaceDetector  framework into Xcode.
Would you please give me a details steps to fix the issue, it's block me for a day ?_? ?

Thanks & Best Regards
Sui
2018-1-26

kesha-antonov, 2018-01-26:
Better to make it optional. I'm definitely will not using this feature right now. I just want to use latest bug fixes.
Is that possible?

suijunqiang, 2018-01-26:
@kesha-antonov  would please help me to add this pod frameworks settings into my Xcode?

Thanks & Best Regards
Sui

suijunqiang, 2018-01-26:
￼
￼

evansherry, 2018-01-26:
@suijunqiang If you do not need to use face recognition, I suggest you set the "react-native-camera" : "0.12.0", not : "git+https://git@github.com/react-native-community/react-native-camera"

sibelius, 2018-01-26:
@aalices how can we make face recognition optional ?

can we do in both ios and android?

jgfidelis, 2018-01-26:
I have found a way to do it in iOS.

But in Android, I tried but could not figure out a good solution because in Java there is no Preprocessor Directives.

aalices, 2018-01-26:
cc @sjchmiela

suijunqiang, 2018-01-26:
@evansherry  I tried 0.12.0, and it will cause more errors like this:

Showing All Errors Only
No such file or directory: '/Users/SJQ/Desktop/dev/rn/percent_sjq/node_modules/react-native-camera/ios/RCT/CameraFocusSquare.m'

what I did:
npm install react-native-camera@0.12.0 --save
npm link react-native-camera@0.12.0

xcode->clean->build

sjchmiela, 2018-01-26:
Exactly like @jgfidelis said, on iOS it's quite easy to make FaceDetector optional by using compiler directives and stubs, like in https://github.com/expo/expo/blob/master/ios/Exponent/Versioned/Core/Api/Components/Camera/EXCamera.h#L7-L11.

On Android I would try to look into something like https://stackoverflow.com/questions/16822895/java-make-a-library-and-import-optional.

jgfidelis, 2018-01-26:
@sjchmiela Seeing the ExpoCamera iOS implementation, if you don't want to import the frameworks and use face detection, than you would simply delete the Optional/FaceDetector folder from the project?

For Android I tried to implement a solution using factories and interfaces but did not finish it, I will look into the projects that your link specified as an example, thanks.

sjchmiela, 2018-01-26:
In Expo there are two ways of building standalone apps: the [more](https://docs.expo.io/versions/latest/guides/detach.html) and [less](https://docs.expo.io/versions/latest/guides/building-standalone-apps.html) configurable one.

The less configurable one packages FaceDetector by default into the application.

The more configurable one allows developer to [specify pods to install, eg. ones provided by Expo](https://docs.expo.io/versions/latest/sdk/payments.html#adding-the-payments-module-on-ios), like Payments or FaceDetector. When the app is detached, there is just no `Optional/{MODULE}` folder. Installing specific adds specific folders.

SamMatthewsIsACommonName, 2018-02-23:
Hi,
Sorry to be a hassle but could someone please suggest where I might be going wrong here?
I've tried to follow the instructions of adding the google frameworks, and I'm still getting build fail with error:
.../node_modules/react-native-camera/ios/FaceDetector/RNFaceDetectorModule.h:9:9: 'GoogleMobileVision/GoogleMobileVision.h' file not found

I've tried following the instructions as closely as I can:

[image]
[image]

Also I've tried restarting xcode, restarting computer, blowing away the derived data folder, cleaning npm modules and cache and re installing them.  I also tried 'chopping out' the Face Detector part of the library, but that didn't work as I have no idea what I'm doing there haha... Any ideas? I'm really at a bit of a loss because I've spent days updating to work with RNCamera on android and it's finally working great with orientation etc...
Thanks in advance for any advice!

maxlang, 2018-03-18:
@SamMatthewsIsACommonName I'm not a react-native or xcode expert by any measure, but here's how I managed to get this working using @kesha-antonov's comment https://github.com/react-native-community/react-native-camera/issues/1151#issuecomment-360246609, but in `RNCamera.xcodeproj`. Hopefully something here can unblock you.

Specifically I changed my Framework Search Paths ***in the `RNCamera.xcodeproj` project*** from `$(SRCROOT)/../../ios` to `$(SRCROOT)/../../../ios` (recursive). Unfortunately this doesn't seem to be a permanent fix since when you install the node module again, framework search paths are overwritten.

After getting it to work I tried messing around with a bunch of options (e.g. checking "copy files if necessary" when adding `RNCamera.xcodeproj` among other things.) And now I can't cause the error anymore, so I'm not sure exactly how to get this working permanently.

I'd recommending looking through your Framework Search Paths and Header Search Paths and making sure everything makes sense. For example, the original Framework Search Paths in `RNCamera.xcodeproj` included `$(SRCROOT)/../../ios` which, the way I included it, resolves to `.../my_project/node_modules/react-native-camera/ios/../../ios`. This in turn resolves to `.../my_project/node_modules/ios` which isn't a valid folder.

[image]

Hope this helps!