react-native-camera/react-native-camera #1152: iOS build failure due to framework search path

tirrorex, 2018-01-23:
### Which implementation are you using

RCTCamera

### Steps to reproduce
Create a project with react-native init (react will be added in libraries
Added react-native-camera to libraries
pod install
Try to build

### Expected behaviour
Project should compile normally

### Actual behaviour
Not compiling.

Problem comes from https://github.com/react-native-community/react-native-camera/commit/8ebeb530c55b6e7559ace7cec3447713e458c235#diff-c35a6eca356820416e0423eb03754a34

Path were added for FRAMEWORK_SEARCH_PATHS, HEADER_SEARCH_PATHS and LIBRARY_SEARCH_PATHS making the compilation fail with error : 
"Argument list too long: recursive header expansion failed at /Users/thibautnoah/Mvpstars/frimousse/node_modules/react-native-camera/ios/../../../ios/Pods/React/node_modules/fsevents/node_modules/tar-pack/test."

Trying to debug this, problem with submitting a PR will be to test that there is no regression for other users.

tirrorex, 2018-01-23:
I changed the title, it wasn't clear that the issue was related to master only

sibelius, 2018-01-23:
@tirrorex check this PR https://github.com/react-native-community/react-native-camera/pull/1153

tirrorex, 2018-01-24:
Yup thanks, i ended up removing RN and Detector also because i didn't want to add more unecessary frameworks to the project (since pod installing google vision is mandatory atm)

nikolas7892, 2018-03-28:
I keep getting the same error.... Argument list too long: recursive header expansion failed at /Users/*USER*/*APP*/node_modules/react-native-camera/ios/../../../ios/build/Index/DataStore/v5/records/1G.

MaheshNandam, 2018-04-05:
I have added Pods at frame relative paths and also [replace master repo at package.json](https://github.com/react-native-community/react-native-camera#how-to-use-master-branch)

but still:
Argument list too long: recursive header expansion failed at ../node_modules/react-native-camera/ios/../../../ios/Pods/React/node_modules/fsevents/node_modules/path-is-absolute.

futurechallenger, 2018-06-02:
@nikolas7892 Hi have u fixed this problem?

arapocket, 2018-06-08:
I'm getting this error as well.

sibelius, 2019-03-18:
check this https://github.com/react-native-community/react-native-camera/issues/1407

closed 2019-03-18 by sibelius