react-native-camera/react-native-camera #1156: Customize Barcode scanner view - Android

ahetawal-p, 2018-01-24:
### Which implementation are you using
RCTCamera 

Is there a way to add a surrounding box around the scanner view. I am looking at the `zxing` examples and see you can achieve something like this 

[image]

Is there a way to get a similar view in this module. Barcode reader without bounds certainly doesnt tell users as where they should focus there camera on.

Thank you,

ludalex, 2018-01-29:
There is #874 that addresses this and more, even though it was recently closed by the author due to the lack of involvement from the repo maintainers to merge his code.

jqn, 2018-02-07:
@ahetawal-p if you need a quick solution you could use this https://github.com/jqn/AC-QRCode-RN with npm or just extract https://github.com/jqn/AC-QRCode-RN/blob/master/lib/QRScanner.js from this library and create your own component.

ahetawal-p, 2018-02-09:
@jqn thanks will take look !

csumrell, 2018-03-02:
I struggled with this for awhile but its actually very simple, just create a visual border and nest it within the RNCamera component

```
import React, { Component } from "react";
import { StyleSheet, View } from "react-native";

export class BarcodeFinder extends Component {
  constructor(props) {
    super(props);
  }

  getSizeStyles() {
    return {
      width: this.props.width,
      height: this.props.height
    };
  }

  render() {
    return (
      <View style={[styles.container]}>
        <View style={[styles.finder, this.getSizeStyles()]}>
          <View
            style={[
              { borderColor: this.props.borderColor },
              styles.topLeftEdge,
              {
                borderLeftWidth: this.props.borderWidth,
                borderTopWidth: this.props.borderWidth
              }
            ]}
          />
          <View
            style={[
              { borderColor: this.props.borderColor },
              styles.topRightEdge,
              {
                borderRightWidth: this.props.borderWidth,
                borderTopWidth: this.props.borderWidth
              }
            ]}
          />
          <View
            style={[
              { borderColor: this.props.borderColor },
              styles.bottomLeftEdge,
              {
                borderLeftWidth: this.props.borderWidth,
                borderBottomWidth: this.props.borderWidth
              }
            ]}
          />
          <View
            style={[
              { borderColor: this.props.borderColor },
              styles.bottomRightEdge,
              {
                borderRightWidth: this.props.borderWidth,
                borderBottomWidth: this.props.borderWidth
              }
            ]}
          />
        </View>
      </View>
    );
  }
}

var styles = StyleSheet.create({
  container: {
    alignItems: "center",
    justifyContent: "center",
    position: "absolute",
    top: 0,
    right: 0,
    bottom: 0,
    left: 0
  },
  finder: {
    alignItems: "center",
    justifyContent: "center"
  },
  topLeftEdge: {
    position: "absolute",
    top: 0,
    left: 0,
    width: 40,
    height: 20
  },
  topRightEdge: {
    position: "absolute",
    top: 0,
    right: 0,
    width: 40,
    height: 20
  },
  bottomLeftEdge: {
    position: "absolute",
    bottom: 0,
    left: 0,
    width: 40,
    height: 20
  },
  bottomRightEdge: {
    position: "absolute",
    bottom: 0,
    right: 0,
    width: 40,
    height: 20
  }
});

```

```
<RNCamera
        ref={cam => {
          this.camera = cam;
        }}
        style={styles.preview}
        onBarCodeRead={data => this.scannedBarcode(data)}
      >
        <BarcodeFinder width={280} height={220} borderColor="red" borderWidth={2} />
      </RNCamera>
```

sibelius, 2018-03-02:
follow this https://github.com/react-native-community/react-native-camera/issues/1156#issuecomment-370035072

closed 2018-03-02 by sibelius

ahetawal-p, 2018-03-02:
The above solution will just show the borders, but will not restrict the scanning to the areas which are not under the border. So the experience would be confusing as to where exactly a user needs to focus the rectangle at. Not sure why this was closed @sibelius

csumrell, 2018-03-02:
That is correct in that it won't limit the scanning areas, but in what case is that going to be issue? 

From my experience users think that the barcode scanner doesn't work if there aren't scan lines...so just give them the fake lines

Spisaczek, 2018-03-06:
Yeah, but limiting scan area should also improve performance.

lucasleelz, 2018-03-29:
Yes,I also think that this is not very good and requires a real scan code area. @sibelius

twxia, 2018-06-08:
Boundary seems need to use below method to get the actual corner coordinates: https://github.com/zxing/zxing/blob/7a64f483fd62afb13f651af655200a6bbfefd1db/core/src/main/java/com/google/zxing/aztec/detector/Detector.java#L235 

So `bounds` parameter in `onBarCodeRead` is same behavior in both iOS (`bounds.origin`) and Android.

sibelius, 2018-06-08:
@twxia can u open another issue and send a pr fixing this?

icxcat, 2019-04-25:
```
		const x = Number(e.bounds.origin.x);
		const y = Number(e.bounds.origin.y);
                 // IOS
		// const width = Number(e.bounds.size.width);
		// const height = Number(e.bounds.size.height);
                // Android
		const width = Number(e.bounds.width);
		const height = Number(e.bounds.height);
		const viewMinX = (winWidth - 267) / 2;
		const viewMinY = 144;
		const viewWidth = 267;
		const viewHeight = 267;
```

wahyuutomoputra, 2019-10-14:
u can try this one https://github.com/shahnawaz/react-native-barcode-mask

VSV6, 2020-09-30:
Thank you so much @wahyuutomoputra ! You saved me from writing the code from scratch! Thanks a lot :)

PR #1824 (2018-10-19): [Android] Fix scanned object bounds (coordinates)
Will Fix #1156 #1503