react-native-camera/react-native-camera #1159: react-native-camera/ios/FaceDetector/RNFaceDetectorUtils.h:12:9:  'GoogleMVDataOutput/GoogleMVDataOutput.h' file not found

suijunqiang, 2018-01-25:
RCTCamera 

### Steps to reproduce
1. rm -rf node_modules/
2.npm install
3.rnpm link
4,Use Xcode clean/build

### Expected behaviour
Xcode throw out errors:

Showing All Errors Only

Build target RCTCamera of project RCTCamera with configuration Debug

CompileC /Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/Objects-normal/x86_64/RNFaceDetectorUtils.o FaceDetector/RNFaceDetectorUtils.m normal x86_64 objective-c com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/SJQ/Desktop/dev/rn/percent_sjq/node_modules/react-native-camera/ios
    export LANG=en_US.US-ASCII
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin"
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x objective-c -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -std=gnu99 -fobjc-arc -fmodules -fmodules-cache-path=/Users/SJQ/Library/Developer/Xcode/DerivedData/ModuleCache -fmodules-prune-interval=86400 -fmodules-prune-after=345600 -fbuild-session-file=/Users/SJQ/Library/Developer/Xcode/DerivedData/ModuleCache/Session.modulevalidation -fmodules-validate-once-per-build-session -Wnon-modular-include-in-framework-module -Werror=non-modular-include-in-framework-module -Wno-trigraphs -fpascal-strings -O0 -Wno-missing-field-initializers -Wno-missing-prototypes -Werror=return-type -Wunreachable-code -Wno-implicit-atomic-properties -Werror=deprecated-objc-isa-usage -Werror=objc-root-class -Wno-arc-repeated-use-of-weak -Wduplicate-method-match -Wno-missing-braces -Wparentheses -Wswitch -Wunused-function -Wno-unused-label -Wno-unused-parameter -Wunused-variable -Wunused-value -Wempty-body -Wuninitialized -Wconditional-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wconstant-conversion -Wint-conversion -Wbool-conversion -Wenum-conversion -Wno-float-conversion -Wno-non-literal-null-conversion -Wno-objc-literal-conversion -Wshorten-64-to-32 -Wpointer-sign -Wno-newline-eof -Wno-selector -Wno-strict-selector-match -Wundeclared-selector -Wno-deprecated-implementations -DDEBUG=1 -DOBJC_OLD_DISPATCH_PROTOTYPES=0 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator11.2.sdk -fasm-blocks -fstrict-aliasing -Wprotocol -Wdeprecated-declarations -mios-simulator-version-min=8.2 -g -Wno-sign-conversion -Wno-infinite-recursion -Wno-comma -Wno-block-capture-autoreleasing -Wno-strict-prototypes -fobjc-abi-version=2 -fobjc-legacy-dispatch -index-store-path /Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Index/DataStore -iquote /Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/RCTCamera-generated-files.hmap -I/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/RCTCamera-own-target-headers.hmap -I/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/RCTCamera-all-target-headers.hmap -iquote /Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/RCTCamera-project-headers.hmap -I/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include -I../../react-native/React -I../../react-native/React/Base -I../../react-native/React/CxxBridge -I../../react-native/React/CxxModule -I../../react-native/React/CxxUtils -I../../react-native/React/DevSupport -I../../react-native/React/Executors -I../../react-native/React/Modules -I../../react-native/React/Profiler -I../../react-native/React/Views -I/Users/SJQ/Desktop/dev/rn/percent_sjq/node_modules/react-native-camera/ios/../../../React -I/Users/SJQ/Desktop/dev/rn/percent_sjq/node_modules/react-native-camera/ios/../../../ios/Frameworks -I/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/DerivedSources/x86_64 -I/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/DerivedSources -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/percent_sjq.app -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/cxxreact -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/double-conversion -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/jschelpers -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/RCTAnimation -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/RCTBaiduMap -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/RCTBEEPickerManager -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/RCTImage -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/React -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/RNDeviceInfo -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/RNSound -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/include/yoga -F/Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Products/Debug-iphonesimulator/percent_sjq.app/PlugIns -F/Users/SJQ/Desktop/dev/rn/percent_sjq/node_modules/react-native-camera/ios/../../../ios/Frameworks -MMD -MT dependencies -MF /Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/Objects-normal/x86_64/RNFaceDetectorUtils.d --serialize-diagnostics /Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/Objects-normal/x86_64/RNFaceDetectorUtils.dia -c /Users/SJQ/Desktop/dev/rn/percent_sjq/node_modules/react-native-camera/ios/FaceDetector/RNFaceDetectorUtils.m -o /Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/Objects-normal/x86_64/RNFaceDetectorUtils.o

In file included from /Users/SJQ/Desktop/dev/rn/percent_sjq/node_modules/react-native-camera/ios/FaceDetector/RNFaceDetectorUtils.m:9:
/Users/SJQ/Desktop/dev/rn/percent_sjq/node_modules/react-native-camera/ios/FaceDetector/RNFaceDetectorUtils.h:12:9: fatal error: 'GoogleMVDataOutput/GoogleMVDataOutput.h' file not found
#import <GoogleMVDataOutput/GoogleMVDataOutput.h>
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.

[... cut ...]
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: can't open file: /Users/SJQ/Library/Developer/Xcode/DerivedData/percent_sjq-cbgtewmquwpcsrbyphxfpctiivzl/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/Objects-normal/x86_64/RNCamera.o (No such file or directory)

### Actual behaviour
It can't be fix, I have try a lot times

### Environment
- **Node.js version**: v8.4.0
- **React Native version**:react-native ^0.46.0
- **React Native platform + platform version**: Xcode 9.2;

### react-native-camera
**Version**: I try it base on last & ^0.12.0

It's run well today before I rm -rf node_modules/

greatdg, 2018-01-26:
try to this link. it works for me. https://github.com/react-native-community/react-native-camera/issues/1151

suijunqiang, 2018-01-26:
@greatdg 
Thanks for your answer,  it is same issue of mine.
But in the post of #1151 I still can not fix my issue,  so would you please give me a hand  if have some experience .

Thanks & Best Regards
Sui
2018-01-26

suijunqiang, 2018-01-26:
There is a work around way of back react-native-camera to 0.12.0, and it able to fix my issue.
so Would you please closed it.

Thanks & Best Regards
Sui

closed 2018-01-26 by sibelius

pohsiu, 2018-03-12:
**No Face Detection steps**
If you do not need it and do not want to install the GMV frameworks, open your app xcode project, on the Project Navigator, expand the RNCamera project, right click on the FaceDetector folder and delete it (move to trash, if you want). If you keep that folder and do not follow the GMV installation setps, your project will not compile.

hungdev, 2018-05-17:
thank @paiou .
as you @pohsiu  say.  after that, clean project by using: comand+ option+shift+k and command + shift + k