react-native-camera/react-native-camera #1160: Failed to resolve errors in android

bhavyasruthi, 2018-01-25:
### Which implementation are you using

RCTCamera 

### Steps to reproduce
Followed all steps in readme file.
### Expected behaviour
Error:Failed to resolve: com.android.support:exifinterface:26.0.2
Error:Failed to resolve: com.github.react-native-community:cameraview:df60b07573

### Actual behaviour
Program should compile 

### Environment
- **Node.js version**:3.10.0
- **React Native version**: 3.10.0
- **React Native platform + platform version**: Developing on 10.13.1 , running on Android emultor

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sibelius, 2018-01-25:
follow Q&A https://github.com/react-native-community/react-native-camera#q--a

closed 2018-01-25 by sibelius

bhavyasruthi, 2018-01-25:
@sibelius I followed Q & A also it dint helped me ! You would not have closed the issue.

sibelius, 2018-01-25:
this Error:Failed to resolve: com.github.react-native-community:cameraview:df60b07573  is a problem with jitpack

u should include exitinterface Error:Failed to resolve: com.android.support:exifinterface:26.0.2

u can ask for more help on expo slack on react-native-camera channel

bhavyasruthi, 2018-02-02:
@sibelius Thank you for your help, I did follow all steps but version updates are missed but your answer helped me to solve it.

actooalizame, 2018-02-05:
I'm having this same issue. Already followed the Q&A guide, but those errors aren't this one.
How did you manage to solve it?

bhavyasruthi, 2018-02-05:
I again went through all those steps, - mainly i added jitpack lines

GarryLaly, 2018-04-28:
Following this step https://medium.com/react-native-zone/error-react-native-camera-com-android-support-exifinterface-26-0-2-7e779366c828 I tested is work.