react-native-camera/react-native-camera #1161: `Camera` no propType for native prop `RNCamera.accessibilityLabel` of native type `String`

emmajam, 2018-01-25:
### Which implementation are you using

RNCamera

### Steps to reproduce
1.     "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera" `npm install` in package.json

2. react-native link react-native-camera
3. react-native run-android OR run in android studio (either or)
4. error pops up

### Expected behaviour
Load app successfully, no errors

### Actual behaviour
Error occurs:
[image]

### Environment
- **Node.js version**: v9.2.1
- **React Native version**: 
react-native-cli: 2.0.1
react-native: 0.43.0
- **React Native platform + platform version**: Android 7.0

### react-native-camera
**Version**: "git+https://git@github.com/react-native-community/react-native-camera"

seblau, 2018-01-27:
Should be fixed by https://github.com/react-native-community/react-native-camera/pull/1165

emmajam, 2018-01-28:
thanks @seblau 🎉

closed 2018-01-28 by emmajam