react-native-camera/react-native-camera #1162: Capture() function takes about 4 to 5 seconds to return something. Sometimes there is no return.

adhenrique, 2018-01-26:
### Which implementation are you using

RNCamera?

### Steps to reproduce
1. Cloned Repo
2. I copied only RNCameraExample folder and changed package.json to latest packages: react, react-native and react-native-camera
3. Added react-navigation package
4. Npm install
5. I created a home screen with a button to navigate to another component, wich contains Camera component
6. I plugged my phone on pc
7. Npm start and in another cmder tab, i run react-native run-android

### Expected behaviour
I put a console.log/alert in the capture() function. When I photograph, I want a "data" return.

### Actual behaviour
I only get the "data" return in the first time I shoot, and it takes about 4 to 5 seconds. In the next attempts I have no return.

### Environment
- **Node.js version**: 8.9.3
- **React Native version**: 0.52.1
- **React Native platform + platform version**: Android 7.0 - Galaxy S6

### react-native-camera
**Version**: master

jgfidelis, 2018-01-26:
RNCameraExample uses RNCamera which does not have the capture method. Are you talking about takePicture(options)?

adhenrique, 2018-01-26:
@jgfidelis yeah, takePicture function!

bvv218, 2018-01-27:
It's been an issue for a long time. Android platform only. 
[768](https://github.com/react-native-community/react-native-camera/issues/768) 
[1000](https://github.com/react-native-community/react-native-camera/issues/1000)

adhenrique, 2018-01-30:
[https://github.com/react-native-community/react-native-camera/issues/908](https://github.com/react-native-community/react-native-camera/issues/908)

ospfranco, 2018-03-29:
I wanted to give you some context on this:

Here is the result of my testing:

```
WITH Default RGB COLOR SPACE and 100% quality

Timing: Starting picture saving 1522320060696
Decoding Done -548
Rotation & Manipulation Done -12
Saving Done -1436

WITH Default RGB 565 and 80% quality

Starting picture saving 1522320335952
Decoding Done -479
Rotation & Manipulation Done -13
Saving Done -1044

Starting picture saving 1522320347745
Decoding Done -475
Rotation & Manipulation Done -8
Saving Done -1067

Starting picture saving 1522320351839
Decoding Done -484
Rotation & Manipulation Done -10
Saving Done -1049

RGB 565 and 100% quality

Starting picture saving 1522320713532
Decoding Done -458
Rotation & Manipulation Done -14
Saving Done -1953

Starting picture saving 1522320718464
Decoding Done -459
Rotation & Manipulation Done -8
Saving Done -1984

Starting picture saving 1522320723406
Decoding Done -493
Rotation & Manipulation Done -11
Saving Done -2048

default RGB and 80% quality
Starting picture saving 1522321728761
Decoding Done -494
Rotation & Manipulation Done -13
Saving Done -957

Starting picture saving 1522321732670
Decoding Done -527
Rotation & Manipulation Done -8
Saving Done -987

Starting picture saving 1522321737130
Decoding Done -523
Rotation & Manipulation Done -8
Saving Done -1010

```

As you can see manipulation (with nothing set) does not take that long, however decoding and saving is taking a lot of time... even when reducing quality to 80% and using a different RGB color space (which is supposed to be faster) does not help the issue.

I tried different approachs to make this faster but it is not really possible, it seems the problem is both decoding/enconding from a byte array to a bitmap and then saving to disk.

I'm not an android expert but there is no way to make this more efficient, when using the default camera API it actually already returns a jpeg image ([link](https://developer.android.com/reference/android/hardware/Camera.html#takePicture(android.hardware.Camera.ShutterCallback,%20android.hardware.Camera.PictureCallback,%20android.hardware.Camera.PictureCallback))) so this seems the way to go for me.

@jgfidelis @sibelius

chcepe, 2018-04-02:
Same issue with Asus Zenfone series

sibelius, 2018-04-02:
Anybody can try to use jpeg instead of convert to bitmap?

wcandillon, 2018-04-02:
@sibelius What are you recommending exactly? I'm tried many options but it always takes 4/5 seconds for `takePictureAsync` to respond.

meiqi1992, 2018-11-15:
Same issue ，how to do it? really need help~