react-native-camera/react-native-camera #1167: User The Camera to Scan , the error：unddefined is not an object（evaluating ‘_react2.propTypes.oneOfType’）

TBoyLi, 2018-01-28:
### Which implementation are you using
``
import Camera from 'react-native-camera'
export  default class Scan extends Component {
    render() {
        return (
            <View style={styles.container}>
                <Camera
                    ref={(cam) => {
                        this.camera = cam;
                    }}
                    onBarCodeRead={this.onBarCodeRead.bind(this)}
                    style={styles.preview}
                    aspect={Camera.constants.Aspect.fill}>
                    <Text style={styles.capture} onPress={this.takePicture.bind(this)}>[CAPTURE]</Text>
                </Camera>
            </View>
        );
    }
}

``
RCTCamera or RNCamera?
   Just use the camera to scan. 
### Steps to reproduce
1. just sen Which implementation are you using
2.
3.

### Expected behaviour
use the camera to scan

### Actual behaviour
unddefined is not an object（evaluating ‘_react2.propTypes.oneOfType’）

### Environment
- **Node.js version**: v8.2.1
- **React Native version**:0.52.2
- **React Native platform + platform version**: iOS 11.2 IPone X, Android 7.0 HuaWei M3, etc

### react-native-camera
**Version**: "react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgcmarins, 2018-01-28:
Hello @TBoyLi. Could you please try to use the latest version? 0.130.
Also consider to try the new camera component, which is `RNCamera`.
Thanks.

TBoyLi, 2018-01-29:
@jgcmarins ok! I'll contact you if I have any other questions. thanks

TBoyLi, 2018-01-29:
@jgcmarins I mean, what's the difference between RNCamera and Camera?

sibelius, 2018-01-29:
RNCamera is based on expo camera module

TBoyLi, 2018-02-04:
@jgcmarins i just write this sentence  import {RNCamera} from 'react-native-camera' and meet this error unddefined is not an object（evaluating ‘_react2.propTypes.oneOfType’）

suhailjain, 2018-02-27:
i am facing the same error but only while deploying app on real android device and not running on simulators. The build succeeds and the bundling also goes well but then the first thing to show up on the device is this error.
Please help

Fconstant, 2018-02-27:
@TBoyLi What is your version of your `react-native-camera` and `react`?

suhailjain, 2018-02-27:
"react": "16.0.0",
 "react-native": "0.51.0",
 "react-native-camera": "^0.6.0",

sibelius, 2018-02-27:
fixed on master and 1.0.0

closed 2018-02-27 by sibelius

Fconstant, 2018-02-27:
Make sure your version of `react-native-camera` is really "master" or "^1":
```
npm list react-native-camera
```

suhailjain, 2018-02-27:
build failing when tried with 1.0.0 .
error: could not resolve all files for configuration ':app:debugCompileClasspath'.
could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.
Searched in the following locations: ........ ............. etc etc

sibelius, 2018-02-27:
check docs

suhailjain, 2018-02-27:
are you pointing towards using RNCamera instead of Camera as of 1.0.0 ? because doing that is also not helping.

Fconstant, 2018-02-27:
@suhailjain [look at the android docs](https://github.com/react-native-community/react-native-camera#android) (Step 6).

suhailjain, 2018-02-28:
after adding jitpack.io, now, the build still succeeds and bundler goes well but this time with a different error : undefined is not an object(evaluating 'CameraManager.Aspect')  coming from ' react-native-camera\src\Camera.js:75:26 '.

Please help

Fconstant, 2018-02-28:
`aspect` and `RNCamera.Contants.Aspect` aren't present on the new version of the camera.
For more details about migration, [see this](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)

suhailjain, 2018-03-01:
Thanks for your help @Fconstant. Appreciate it.

arapocket, 2018-03-06:
@Fconstant I've followed the steps in the migration section (which by the way, never mentions anything about Aspect) to no avail.

Fconstant, 2018-03-07:
@arapocket Yes, I've noticed this too. You can send a PR for `migration.md` file adding this section.

About your migration problem, please send me some details so I can help you.