react-native-camera/react-native-camera #1171: 重大BUG

wabgy, 2018-01-29:
### Which implementation are you using

RCTCamera or RNCamera?

RNCamera

### Steps to reproduce
1.
2.
3.

### Expected behaviour
react-native-camera:unspecified

### Actual behaviour

A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:exifinterface:26.0.2.
     Searched in the following locations:
         file:/E:/android/android-sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.pom
         file:/E:/android/android-sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
         file:/F:/ReactNativeWorkspace/QRCodeTest/android/sdk-manager/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
     Required by:
         QRCodeTest:app:unspecified > QRCodeTest:react-native-camera:unspecified
   > Could not find com.github.react-native-community:cameraview:df60b07573.
     Searched in the following locations:
         file:/E:/android/android-sdk/extras/android/m2repository/com/github/react-native-community/cameraview/df60b07573/cameraview-df60b07573.pom
         file:/E:/android/android-sdk/extras/android/m2repository/com/github/react-native-community/cameraview/df60b07573/cameraview-df60b07573.jar
         file:/F:/ReactNativeWorkspace/QRCodeTest/android/sdk-manager/com/github/react-native-community/cameraview/df60b07573/cameraview-df60b07573.jar
     Required by:
         QRCodeTest:app:unspecified > QRCodeTest:react-native-camera:unspecified

### Environment
- **Node.js version**:
- **React Native version**:0.48.2
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**: npm version

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

sibelius, 2018-01-29:
Check Q & A on reamed

closed 2018-01-29 by sibelius

haofeng123, 2018-01-30:
i have this question +1