react-native-camera/react-native-camera #1172: The camera capture is not filled entirely into <Camera> component

swingywc, 2018-01-29:
### Which implementation are you using

RCTCamera or RNCamera?  

**I am using RNCamera.**

### Steps to reproduce
    import React, { Component } from 'react';
    import { Modal, View, Dimensions } from 'react-native';
    import Camera from 'react-native-camera';

    const { width, height } = Dimensions.get('window');

    export default class QrCameraModal extends Component {
      ...
      ...

      render() {
        return (
          <Modal visible={this.state.modalVisible} animationType={'slide'} onRequestClose={ () => this.closeModal() }>
            <View style={{flex: 1}}>
              <Camera
                onBarCodeRead={this.onQRCodeRead.bind(this)}
                aspect="stretch"
                style={{width: width, height: height}}>
              </Camera>
            </View>
          </Modal>
        );
      }
    }

### Expected behaviour
Aspect camera capture should fill the <Camera> component (i.e. the entire modal).

### Actual behaviour

In my device, the capture view is not filled inside the <Camera> component.  

[image]

However in simulator, I am sure the <Camera> component is already fullscreen.  

[image]

### Environment
- **Node.js version**: 8.9.1
- **React Native version**: 0.52.0

### react-native-camera
**Version**: ^0.10.0

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgcmarins, 2018-01-30:
hey @swingywc would you mind to try version 0.13.0 and `RNCamera` component? thanks

swingywc, 2018-01-30:
hi @jgcmarins , first of all thank you for your kindly help. I have followed your instruction to install v0.13.0 and use `import { RNCamera } from 'react-native-camera';`. However, I am facing the following problem now.

`Invariant Violation: Native component for 'RNCamera' does not exist`  

I have searched around the issue and documentation, so far what I got is RNCamera is referenced to Expo's Camera component. Nevertheless, I am not using Expo to test my application. I am using XCode to build app into my device and `react-native run-ios` to build app into simulator. Can you give me some suggestion on how to fix this error? Thank you so much.

jgcmarins, 2018-01-30:
I think this was also fixed on `master`. Can you try it on `master` branch?
Follow this: https://github.com/react-native-community/react-native-camera#experimental
Also make sure you are linking it correctly.

swingywc, 2018-01-30:
Oh I found that using `<Camera>` component in v0.13.0 fix the bottom black bar problem. Thank you so much @jgcmarins . You save my day. ;)

closed 2018-01-30 by swingywc