react-native-camera/react-native-camera #1174: RNCamera: Android Portrait Orientation Upside Down.

DevarakondaV, 2018-01-30:
### Which implementation are you using
Using RNCamera Implementation

### Steps to reproduce
1. Take Picture from front camera.
2. Save using CameraRoll API.

### Expected behaviour
When I take a picture, store it using the CameraRoll API and look inside my Gallerias, the image should be right side up.

### Actual behaviour
What actually happens is that the image gets flipped upside down. 

### Environment
- **Node.js version**:v8.9.3
- **React Native version**: 0.51.0
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
Using master version. 

###Possible Solution
We resolved this issue but decided to make a post for others that might be experiencing the same thing. Not sure if this is the correct implementation, so would need someone to verify. The issue for us seems to be the following line:

Line55: mutableImage.mirrorImage()

inside: react-native-camera/android/src/main/java/org/reactnative/camera/tasks/ResolveTakenPictureAsyncTask.java

AlgoTrader, 2018-01-31:
I have the same problem. Sometimes the shots are flipped horizontally, but sometimes vertically. It depends on device. Preview is always OK but all the shots are always mirrored

DevarakondaV, 2018-01-31:
@AlgoTrader 

Have you tried the possible solution I suggested above? 

Also, I just rechecked the file I listed above and it seems there are some new updates made to the master. I haven't tested them but they might solve this issue.

monolithed, 2018-02-05:
The same problem. Is there any way to fix that?

Brioni, 2018-02-06:
//mutableImage.mirrorImage();
Fix it

jgfidelis, 2018-02-06:
Is this still happening on master?

This commit: https://github.com/react-native-community/react-native-camera/commit/91a560ac1cc638ee8cd2f040938d93364b875512

should have fixed image being mirrored.

sibelius, 2018-02-06:
fixed

closed 2018-02-06 by sibelius