react-native-camera/react-native-camera #1175: Need help for creating a barcode scan mask for RNCamera

changx03, 2018-01-30:
How to add a mask on top of the react-native-camera?

I'm building the UI for a React Native QRCode scanner app using react-native-camera.

The overlay mask on top of the camera should be in light grey color, but the middle part must keep transparent (see-through).

But when I change the backgroundColor on my outer mask, it seems also affect the center part. I mean, of course, it is behind its child view.

```
<Camera
  ref={cam => {
    this.camera = cam;
  }}
  onBarCodeRead={this._onBarCodeRead}
  style={styles.cameraView}
  aspect={Camera.constants.Aspect.fill}
  playSoundOnCapture
>
  <View
    style={{
      position: 'absolute',
      top: 0,
      left: 0,
      width: '100%',
      height: '100%',
      backgroundColor: 'rgba(0.2, 0.2, 0.2, 0.2)',
      alignItems: 'center',
      justifyContent: 'space-around',
    }}
  >
    <View
      style={{
        width: 300,
        height: 300,
        backgroundColor: 'transparent',
        borderColor: 'white',
        borderWidth: 1,
      }}
    />
  </View>
</Camera>
```
I posted a screen shot on [stackoverflow](https://stackoverflow.com/questions/48510350/react-native-camera-with-a-transparent-view-for-barcode-scanner-mask)

I know this is my problem. Thank you so many for your help.

### Environment
"react-native-camera": "^0.13.0",
"react-native": "0.52.1",
"react": "16.2.0",

jgcmarins, 2018-01-30:
Maybe you can follow this example: https://github.com/react-native-community/react-native-camera/tree/master/RNCameraExample

changx03, 2018-01-30:
Don't think the example helps my problem. I basically need to add a frame on top of the camera window, so the users feel like they need to focus on the middle part of scanning the barcode. The example only added overlay down the bottom which is much easier to do.

changx03, 2018-02-01:
I finally managed to create a mask by using multiple view components and computing width and height at runtime.
https://stackoverflow.com/questions/48510350/react-native-camera-with-a-transparent-view-for-barcode-scanner-mask/48555424#48555424

closed 2018-02-01 by changx03