react-native-camera/react-native-camera #1176: Couldn't capture photo using Xiaomi phones

khf118, 2018-01-30:
### Which implementation are you using

RCTCamera or RNCamera?
RCTCamera

### Expected behaviour
The camera opens successfully but once I click to take the picture, the camera freezes and I get the Couldn't take picture error in the log instead of receiving the callback. The app works correctly on different phones but I'm having this issue only with two xiaomi phones: Redmi x4 and Redmi Node 4

### Environment
- **Node.js version**:
- **React Native version**: 0.50
- **React Native platform + platform version**: Android 7.1, Android 7.0 etc

### react-native-camera
**Version**: 0.13.0

isnifer, 2018-05-04:
Log from pidcat:
```
 OMXClient  I  MuxOMX ctor
       art  I  Background partial concurrent mark sweep GC freed 4319(569KB) AllocSpace objects, 30(20MB) L
              OS objects, 20% free, 63MB/79MB, paused 3.746ms total 141.513ms
 OMXClient  I  MuxOMX ctor
    Camera  D  app passed NULL surface
AudioTrack  W  AUDIO_OUTPUT_FLAG_FAST denied by client; transfer 4, track 44100 Hz, output 48000 Hz
```

sibelius, 2018-05-04:
get log from adb logcat instead

and provide the full stacktrace

isnifer, 2018-05-04:
@sibelius 
```
05-04 21:16:13.890  1330  1366 E audio_hw_primary: start_dsm_feedback_process: pcm device id 38
05-04 21:16:13.893   756 11504 I mm-camera: <IFACE >< INFO> 382: iface_ispif_get_cfg_params_from_hw_streams: interace 0 active count = 2, skip stop ispif
05-04 21:16:13.893   756 11504 I mm-camera: <IFACE >< INFO> 382: iface_ispif_get_cfg_params_from_hw_streams: interace 0 active count = 2, skip stop ispif
05-04 21:16:13.893   756 11504 I mm-camera: <IFACE >< INFO> 382: iface_ispif_get_cfg_params_from_hw_streams: interace 0 active count = 1, skip stop ispif
05-04 21:16:13.893   756 11504 I mm-camera: <IFACE >< INFO> 382: iface_ispif_get_cfg_params_from_hw_streams: interace 0 active count = 1, skip stop ispif
05-04 21:16:13.895   756 11616 E mm-camera: <IFACE ><ERROR> 693: iface_queue_buf: iface_queue_buf: error, do not have VFE fd
05-04 21:16:13.895   756 11616 E mm-camera: <IFACE ><ERROR> 7812: iface_util_buf_divert_notify: iface_util_buf_divert_notify: queue back buffer failed!hw stream id 9000f, bufq handle bc0007, buf_idx 0 rc = -1
05-04 21:16:13.895   756 11504 I mm-camera: <IFACE >< INFO> 2399: iface_streamoff: session_id = 2, active_streams = 0
05-04 21:16:13.895   756 11504 I mm-camera: <IFACE >< INFO> 2481: iface_streamoff: X
05-04 21:16:13.895   756 11521 I mm-camera: <IFACE >< INFO> 1643: iface_streamoff_to_thread: iface_streamoff_to_thread: X, session id = 2, user stream id = 3
05-04 21:16:13.895   756 11521 I mm-camera: <MCT   >< INFO> 3974: mct_pipeline_process_set:  STREAM-OFF success for stream 0x20003, stream type=3
05-04 21:16:13.896   756 11521 I mm-camera: <MCT   >< INFO> 3814: mct_pipeline_process_set: command=8000009 value=-1
05-04 21:16:13.896   756 11521 I mm-camera: <MCT   >< INFO> 3959: mct_pipeline_process_set:  STREAM-OFF on stream 0x20002 stream type=2
05-04 21:16:13.902   756 11521 I mm-camera: <CPP   >< INFO> 3927: cpp_module_handle_streamoff_event: info: doing stream-off for identity 0x20002
05-04 21:16:13.903   756 11506 I mm-camera: <CPP   >< INFO> 359: cpp_hardware_set_clock: Set clock 465000000 BW avg 16103807 BW inst 16103807
05-04 21:16:13.903   756 11521 I mm-camera: <CPP   >< INFO> 4112: cpp_module_handle_streamoff_event: info: stream-off done for identity 0x20002
05-04 21:16:13.904   756 11521 I mm-camera: <IFACE >< INFO> 1617: iface_streamoff_to_thread: iface_streamoff_to_thread: E, session id = 2, user stream id = 2
05-04 21:16:13.904   756 11504 I mm-camera: <IFACE >< INFO> 2209: iface_streamoff: E, session_id 2 mct_stream_id = 2
05-04 21:16:13.904   756 11504 I mm-camera: <IFACE >< INFO> 10984: iface_util_get_user_streams_by_bundle: skip streamoff! left 0 bundled streamoff, mct stream id 2total bundled stream num = 2
05-04 21:16:13.905   756 11504 I mm-camera: <IFACE >< INFO> 2418: iface_streamoff: session id = 0x2,Delta between vfe irqs 0 sec 0 usec
05-04 21:16:13.906  1330  1366 D msm8916_platform: platform_can_split_snd_device: snd_device(2) num devices(0) new_snd_devices(0)
05-04 21:16:13.906  1330  1366 D audio_hw_primary: enable_snd_device: snd_device(2: speaker)
05-04 21:16:13.906  1330  1366 I soundtrigger: audio_extn_sound_trigger_update_device_status: device 0x2 of type 0 for Event 1, with Raise=0
05-04 21:16:13.907   756 11504 I mm-camera: <IFACE >< INFO> 10029: iface_util_release_resource: camif_cnt= 1, rdi_cnt= 0, used mask 10001
05-04 21:16:13.909  1330  1366 D aud_dev_arbi: acquireOneDevice: device owner for device 2 is 1
05-04 21:16:13.909   756 11504 I mm-camera: <IFACE >< INFO> 10029: iface_util_release_resource: camif_cnt= 0, rdi_cnt= 0, used mask 0
05-04 21:16:13.909  1330  1366 I audio_hw_primary: select_devices Selected apptype: 69937
05-04 21:16:13.909  1330  1366 D soundtrigger: audio_extn_sound_trigger_update_stream_status: uc_id 1 of type 0 for Event 3, with Raise=0
05-04 21:16:13.910   756 11504 I mm-camera: <IFACE >< INFO> 10029: iface_util_release_resource: camif_cnt= 0, rdi_cnt= 0, used mask 0
05-04 21:16:13.910  1330  1366 I audio_hw_utils: audio_extn_utils_send_app_type_cfg PLAYBACK app_type = 69937, acdb_dev_id = 1058, sample_rate = 48000
05-04 21:16:13.911   756 11504 I mm-camera: <IFACE >< INFO> 2476: iface_streamoff: all bundled stream streamoff, clear hal bundled mask
05-04 21:16:13.911  1330  1366 D msm8916_platform: platform_can_split_snd_device: snd_device(2) num devices(1) new_snd_devices(-223345648)
05-04 21:16:13.911   756 11504 I mm-camera: <IFACE >< INFO> 2481: iface_streamoff: X
05-04 21:16:13.911  1330  1366 V msm8916_platform: platform_send_audio_calibration: sending audio calibration for snd_device(2) acdb_id(1058)
05-04 21:16:13.911   756 11521 I mm-camera: <IFACE >< INFO> 1643: iface_streamoff_to_thread: iface_streamoff_to_thread: X, session id = 2, user stream id = 2
05-04 21:16:13.911  1330  1366 D ACDB-LOADER: ACDB -> send_audio_cal, acdb_id = 1058, path = 0, app id = 0x11131, sample rate = 48000
05-04 21:16:13.911  1330  1366 D ACDB-LOADER: ACDB -> send_asm_topology
05-04 21:16:13.912   756 11521 I mm-camera: <MCT   >< INFO> 177: stop_sof_check_thread: Stopping SOF timeout thread session =2
05-04 21:16:13.912   756 11620 I mm-camera: <MCT   >< INFO> 114: mct_bus_sof_thread_run: Closing SOF tracker thread
05-04 21:16:13.912  1330  1366 D ACDB-LOADER: ACDB -> ACDB_CMD_GET_AUDPROC_STREAM_TOPOLOGY_ID
05-04 21:16:13.913   756 11521 I mm-camera: <MCT   >< INFO> 3974: mct_pipeline_process_set:  STREAM-OFF success for stream 0x20002, stream type=2
[... cut ...]
05-04 21:16:13.941   756 11506 I mm-camera: <CPP   >< INFO> 359: cpp_hardware_set_clock: Set clock 465000000 BW avg 16103807 BW inst 16103807
05-04 21:16:13.944  1331 11624 I QCamera : <MCI><INFO> mm_channel_process_stream_buf: 543: Super Buffer received, pending_cnt=0 queue cnt = 0 expected = 0
05-04 21:16:13.944  1331 11624 I QCamera : <MCI><INFO> mm_channel_process_stream_buf: 629: node=0 notify=1 cnt=0 stop=0 manu=0 call=0
05-04 21:16:13.952  1330  1366 D audio_hw_primary: start_output_stream: exit
```

isnifer, 2018-05-07:
Just tested on 1.1.2 — the same result

Mathieuduverney, 2018-10-11:
Still not working on 1.3.0 

No updates ??