react-native-camera/react-native-camera #1177: onFacesDetected Crash (iOS)

ma-pe, 2018-01-30:
### Which implementation are you using

RNCamera

### Steps to reproduce
1. add the onFacesDetected-Listener to an empty function

### Expected behaviour
function is called whenever a face is detected

### Actual behaviour
1. the function is actually called once, when the camera is mounted. the given parameter is: `{"type":"face","faces":[],"target":125}`
2. the app enters instantly an unresponsive state, xcode shows an error in binary code: `Thread 3: EXC_BAD_ACCESS (code=1, address=0x0)`
Console says:
```
2018-01-30 12:34:34.354686+0100 abc[3907:1994192] Instantiated <GMVFaceDetector: 0x1c028e6f0>
2018-01-30 12:34:34.385077+0100 abc[3907:1994192] Failed to locate required resources. Please verify the reosurces bundle is imported.
```

### Environment
- **Node.js version**: v8.9.3
- **React Native version**: 0.51.0
- **React Native platform + platform version**: iOS 11.2.5

### react-native-camera
**Version**: "master"

jgfidelis, 2018-01-30:
Did you copy everything from the downloaded frameworks to your xcode project? Looks like it is not finding one of the resources of a framework. Like the FaceDetector framework has a folder called resource, is it imported into your project?

ma-pe, 2018-01-30:
I had only copied the frameworks.

That did the trick. Thank you!

closed 2018-01-30 by jgfidelis