react-native-camera/react-native-camera #1178: Footprint of Google Mobile Vision on iOS

ma-pe, 2018-01-30:
Embedding the Frameworks for Google Mobile Vision adds a footprint of around 25 mb to my .ipa-file. 

Is there any chance this will be considerably less in the (near) future?

sibelius, 2018-01-30:
this is only needed if u want face detection

you should ask google about it

ma-pe, 2018-01-30:
I do want face detection, but the footprint is immense. 

Following the setup-instructions I added 8 frameworks in order to achieve the face detection. I was just questioning myself, if all are necessary (e.g. the address book framework). :-)

sibelius, 2018-01-30:
https://developers.google.com/vision/ios/getting-started

jgfidelis, 2018-01-30:
Unfortunately, there is nothing we can do about it. Google Mobile Vision uses other Google frameworks and at least one of them uses AddressBook (Google utilities I believe). I tried to remove as much as possible, but these are the minimum requirements for the project to compile correctly.

A solution would be to find a face detection framework other than Google's that is lighter and provides the same functionality.

closed 2018-01-30 by sibelius

ma-pe, 2018-02-01:
@jgcmarins Thanks for clarifying.