react-native-camera/react-native-camera #1183: Could not find com.android.support:exifinterface:26.0.2

sergiofraile, 2018-01-31:
### Which implementation are you using

RCTCamera

### Steps to reproduce
1. I upgraded my app from RN 0.45 to 0.52.2. Camera was working fine.
2. Had to upgrade the camera from "0.9.4" to "0.13.0"
3. It won't build for Android. Works fine for iOS.
4. Tried the Q&A section from the Readme file, issue persists. I wonder if the fact that the library was already linked means some extra manual upgrading.

### Expected behaviour
Build succesful

### Actual behaviour
FAILURE: Build failed with an exception.
Shows the following error:

> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:exifinterface:26.0.2.
     Searched in the following locations:
         file:/Users/{...}/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.pom
         file:/Users/{...}/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
         file:/Users/{...}/Documents/Projects/rn-project/android/sdk-manager/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
     Required by:
         project :app > project :react-native-camera

### Environment
- **Node.js version**: 7.2.1
- **React Native version**: 0.52.2
- **React Native platform + platform version**: Android 7.1.1

### react-native-camera
**Version**: 0.13.0

sibelius, 2018-01-31:
Follow README instructions

closed 2018-01-31 by sibelius

MichaelDanielTom, 2018-02-02:
@sergiofraile Did you end up finding out what was wrong? I've tried upgrading my own project's compileSdkVerison to 26.0.2 but to no avail :/

Any help is appreciated!

sergiofraile, 2018-02-05:
I was able to fix it in the end but nothing much related to the instructions there. It turns out that upgrading my gradle version fixed some of the issues with the dependencies. Not sure if I would have been able to change any other setting instead that would have fixed the issue.

I believe that the version of gradle I upgraded too was able to discover the google() parameter in the dependencies. Just dropping it in here in case someone find it useful. Since I'm not a native android dev, all the gradle system is a bit unknown to me.

hungdev, 2018-02-09:
i get the same error with version: "react-native-camera": "^0.13.0"
i add maven { url "https://maven.google.com" } as guide @MichaelDanielTom https://stackoverflow.com/questions/45357000/failed-to-resolve-com-android-supportappcompat-v726-0-0/45481709#45481709
After that, i get issue: https://github.com/react-native-community/react-native-camera/issues/1086
i resolved by adding maven { url "https://jitpack.io" } from readme.md:
step 6: Add jitpack to android/build.gradle
```
allprojects {
    repositories {
        maven { url "https://jitpack.io" }
    }
}
```
I ended up also running into https://github.com/react-native-community/react-native-camera#meta-data-android-26
and everything worked like @MichaelDanielTom . Thanks @MichaelDanielTom

GarryLaly, 2018-04-28:
Following this step https://medium.com/react-native-zone/error-react-native-camera-com-android-support-exifinterface-26-0-2-7e779366c828 I tested is work.