react-native-camera/react-native-camera #1184: Migration guide from RCTCamera to RNCamera

sibelius, 2018-01-31:
We are just moving to RNCamera is 2 huge production apps at @entria

We plan to deprecated RCTCamera soon, so we need a migration guide to move to RNCamera

Can somebody write a `.md` about it?

AlgoTrader, 2018-01-31:
I have tried RNCamera on Android. The results are very frustrating.

On Honor9 device the back camera shots are flipped vertically, making all the texts mirrored
The same on front Camera

On Samsung J3 the front camera shots are flipped horizontally (making head in the bottom of shot). with the back camera app crashes.

I have the Samsung J3 also crashed with the old RTCCamera though

sibelius, 2018-01-31:
@AlgoTrader can u try latest master

@jgfidelis improve the code a bit yesterday

jgfidelis, 2018-01-31:
@AlgoTrader the image mirroring should be fixed on latest master. If you have more crashes, post them here so we can debug them, please.

AlgoTrader, 2018-01-31:
I have found the crash reason. The old good "largeHeap" on devices with low memory and great camera

Fconstant, 2018-02-28:
Done by [this commits](https://github.com/react-native-community/react-native-camera/commits/master/docs/migration.md)

closed 2018-02-28 by Fconstant

sibelius, 2018-02-28:
@AlgoTrader could u largeHeap to docs?