react-native-camera/react-native-camera #1188: RNCamera - requestPermissions on componentWillMount

hawkup, 2018-01-31:
From this [line](https://github.com/react-native-community/react-native-camera/blob/d14283cdab16e0d8e9fa8820e156028dffb0c77f/src/RNCamera.js#L261) on RNCamera.js
```JS
  async componentWillMount() {
    const hasVideoAndAudio = true; //TODO implement capture mode for camera / video like RCTCamera. Now, infer always video.
    const isAuthorized = await requestPermissions(hasVideoAndAudio, CameraManager, Platform.OS, this.props.permissionDialogTitle, this.props.permissionDialogMessage);
    this.setState({ isAuthorized, isAuthorizationChecked: true });
  }
```

Could **hasVideoAndAudio** change to use the flag props for request permission audio and video instead to use **captureMode** and **captureAudio** props ?

jgfidelis, 2018-01-31:
Hey, this is something I am planning to make better.

My idea, I believe is the same as your suggestion, is in RNCamera to have a flag "hasAudio" as prop. If yes, we will ask for video and audio. If no, we will ask for video only. Is that you idea as well?

hawkup, 2018-01-31:
Yes, I think like you. But when called **recordAsync** it has **mute** options (reference https://docs.expo.io/versions/latest/sdk/camera.html#recordasync).
Should it be the same value ?

closed 2018-02-02 by hawkup

newlifepop, 2018-02-06:
Hi, just a question, I've been working on it for quite a while now but couldn't get it solved. Also asked here and on SO. I have no problems when capturing a photo, but when trying to record videos, videos are not being recorded. Could you maybe share the part of your code where to take a video? Thanks!