react-native-camera/react-native-camera #1199: RNFaceEncoder.m compile error

condorman, 2018-02-01:
### Steps to reproduce
1. react-native init testApp
2. cd testApp
3. npm install -save git+https://git@github.com/react-native-community/react-native-camera
    (after installation react native package is removed)
4. npm install (so react-native package is reinstalled)
5. react-native link react-native-camera
6. react-native run-ios

### Actual behaviour
The following commands produced analyzer issues:
	Analyze /Users/condorman/Documents/interview/node_modules/react-native/ReactCommon/yoga/yoga/Yoga.cpp
	Analyze Base/RCTModuleMethod.mm
(2 commands with analyzer issues)

The following build commands failed:
	CompileC /Users/condorman/Documents/interview/ios/build/Build/Intermediates.noindex/RCTCamera.build/Debug-iphonesimulator/RCTCamera.build/Objects-normal/x86_64/RNFaceEncoder.o FaceDetector/RNFaceEncoder.m normal x86_64 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)

### Environment
- MacOs 10.13
- React native: 0.52.2
- Iphone 6 (emulator), compiler version 11.2

### react-native-camera
**Version**: Master

sibelius, 2018-02-01:
if u want to use FaceDetector you need to link google vision frameworks

if u don't want to you can use a postinstall script like this:

### package.json
```
"postinstall": "./scripts/post.sh",
```
### post.sh
```sh
#!/bin/bash
rm -rf node_modules/react-native-camera/ios/FaceDetector
cp node_modules/react-native-camera/postinstall_project/projectWithoutFaceDetection.pbxproj node_modules/react-native-camera/ios/RCTCamera.xcodeproj/project.pbxproj
```

jgfidelis, 2018-02-01:
Or you can open your app's xcode project, click on Libraries, expand the RCTCamera project, right click on FaceDetector folder and delete (move to trash if you want to).

condorman, 2018-02-01:
work fine! thank you!

closed 2018-02-01 by sibelius

stevelacy, 2018-02-26:
As of Feb 26 with the 1.0.0 release the name of the folders have changed, I modified the `postinstall.sh` file to be:

```sh
#!/bin/bash

rm -rf node_modules/react-native-camera/ios/FaceDetector
cp node_modules/react-native-camera/postinstall_project/projectWithoutFaceDetection.pbxproj node_modules/react-native-camera/ios/RNCamera.xcodeproj/project.pbxproj

```

laurent22, 2019-06-14:
I'm having the same error and none of the solutions mentioned here work (there's no longer a postinstall project for instance). Is there an official way to make this work?

sibelius, 2019-06-14:
use cocoapods