react-native-camera/react-native-camera #1200: Build fails on 0.13 veriosn (works on 0.12)

niksajanjic, 2018-02-01:
### Steps to reproduce
1. npm install react-native-camera --save
2. react-native link react-native-camera
3. react-native run-android

### Expected behaviour
Build should succeed

### Actual behaviour
Build fails with error:
```
* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:exifinterface:26.0.2.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
     Required by:
         Glocaleze:app:unspecified > Glocaleze:react-native-camera:unspecified
   > Could not find com.github.react-native-community:cameraview:df60b07573.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/github/react-native-community/cameraview/df60b07573/cameraview-df60b07573.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/github/react-native-community/cameraview/df60b07573/cameraview-df60b07573.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/github/react-native-community/cameraview/df60b07573/cameraview-df60b07573.jar
     Required by:
         Glocaleze:app:unspecified > Glocaleze:react-native-camera:unspecified
```

### Environment
- node 9.4
- RN 0.52.1
- Nexus 6 real device, Android 7.1.1

### react-native-camera
0.13 (but same is on master)

###  What I also tried to fix the problem:

I added:
```url "https://jitpack.io"```
to my android/build.gradle which resulted in a slightly different error:
```
* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:exifinterface:26.0.2.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
     Required by:
         Glocaleze:app:unspecified > Glocaleze:react-native-camera:unspecified
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         Glocaleze:app:unspecified > com.android.support:appcompat-v7:23.0.1
         Glocaleze:app:unspecified > com.facebook.react:react-native:0.20.1 > com.android.support:recyclerview-v7:23.0.1
   > Could not find com.android.support:support-annotations:26.0.1.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
     Required by:
         Glocaleze:app:unspecified > Glocaleze:react-native-camera:unspecified > com.github.react-native-community:cameraview:df60b07573
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         Glocaleze:app:unspecified > Glocaleze:react-native-camera:unspecified > com.github.react-native-community:cameraview:df60b07573
   > Could not find com.android.support:support-annotations:26.0.1.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/support-annotations/26.0.1/support-annotations-26.0.1.jar
     Required by:
         Glocaleze:app:unspecified > com.facebook.react:react-native:0.20.1 > com.android.support:recyclerview-v7:23.0.1
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         Glocaleze:app:unspecified > Glocaleze:react-native-camera:unspecified > com.google.android.gms:play-services-vision:11.0.4 > com.google.android.gms:play-services-basement:11.0.4
   > Could not find com.android.support:support-v4:26.0.1.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/support-v4/26.0.1/support-v4-26.0.1.jar
     Required by:
         Glocaleze:app:unspecified > com.facebook.react:react-native:0.20.1 > com.facebook.fresco:fresco:0.8.1 > com.facebook.fresco:imagepipeline:0.8.1
         Glocaleze:app:unspecified > com.facebook.react:react-native:0.20.1 > com.facebook.fresco:fresco:0.8.1 > com.facebook.fresco:drawee:0.8.1
```
I also tried adding these https://github.com/react-native-community/react-native-camera#q--a
and I also tried changing `compileSdkVersion 26` and `buildToolsVersion "26.0.1"`. All of that helped to get rid of all errors except the first one:
```
* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:exifinterface:26.0.2.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/exifinterface/26.0.2/exifinterface-26.0.2.jar
     Required by:
         Glocaleze:app:unspecified > Glocaleze:react-native-camera:unspecified
```

I have multiple versions of Android SDK Build-Tools installed:
23.0.1
25.0.2
26.0.1
26.0.2
27.0.3

But all of lowest and highest one got installed automatically through build process.

sibelius, 2018-02-01:
check readme Q & A

closed 2018-02-01 by sibelius

niksajanjic, 2018-02-01:
@sibelius I explained up there that I tried that and it only resulted in slightly different errors, but the error: `Could not find com.android.support:exifinterface:26.0.2` was never resolved

niksajanjic, 2018-02-01:
@sibelius I was able to fix this issue:
```
* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.android.support:exifinterface:26.0.1.
     Searched in the following locations:
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.1/exifinterface-26.0.1.pom
         file:/Users/niksajanjic/Library/Android/sdk/extras/android/m2repository/com/android/support/exifinterface/26.0.1/exifinterface-26.0.1.jar
         file:/Users/niksajanjic/Projects/Glocaleze/glocaleze-native/Glocaleze/android/sdk-manager/com/android/support/exifinterface/26.0.1/exifinterface-26.0.1.jar
     Required by:
         Glocaleze:app:unspecified
```

So, for some reason, I don't have `exifinterface` of version 26.0.1. The latest version I have installed is 26.0.0-alpha1. Therefore I just changed this line in android/app/build.gradle:
```
compile ('com.android.support:exifinterface:26.0.0-alpha1') {
    force = true;
}
```

Do you know by any chance why I don't have that specific version installed and I only have 25.1.0, 25.1.1, 25.2.0, 25.3.0, 25.3.1 and 26.0.0-alpha1?

I have Android SDK Build-Tools of version 26.0.2 (and 27.0.3) so I assumed that means it should cover that module too.

actooalizame, 2018-02-05:
I'm running into this same issue. How did you get to know which version of exifinterface is installed?

SirNeuman, 2018-02-12:
I have the same issue as you @niksajanjic. For some reason i only have the alpha version installed

SirNeuman, 2018-02-12:
Ok. I figured out why it wasn't installing 26.0.1. I needed to include in `android/build.gradle`:

```
allprojects {
        repositories {
            jcenter()
            maven {
                url "https://maven.google.com"
            }
        }
}
```

Shubham-Sinha, 2018-02-13:
Can you guys please share your ``build.gradle`` configurations and ``package.json``. I've tried all permutation and combinations of steps to follow, but I'm unable to make it run

SirNeuman, 2018-02-13:
```
buildscript {
    repositories {
        jcenter()
        mavenCentral()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.3.1'
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven { url "https://jitpack.io" }
        maven {
            url "https://maven.google.com"
        }
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
    }
}
```

@Shubham-Sinha this is my `build.gradle`... I kind of forget what's relevant to this issue other then the `maven.google.com` addition. There's also other stuff in my `build.gradle` file that I know is not relevant to the issue. I also don't think there's anything special about my `package.json` file.

niksajanjic, 2018-02-13:
### build.gradle
```
// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    repositories {
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.2.3'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven {
            url "https://jitpack.io"
        }
    }
}

```

### package.json
```
{
  "name": "test",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "debugger": "open \"rndebugger://set-debugger-loc?host=localhost&port=8081\"",
    "test": "jest"
  },
  "dependencies": {
    "lodash.camelcase": "^4.3.0",
    "lodash.get": "^4.4.2",
    "prop-types": "^15.6.0",
    "react": "16.2.0",
    "react-native": "0.53.0",
    "react-native-camera": "^0.13.0",
    "react-native-config": "^0.11.5",
    "react-native-storage": "^0.2.2",
    "react-navigation": "^1.0.0",
    "react-navigation-redux-helpers": "^1.0.1",
    "react-redux": "^5.0.6",
    "redux": "^3.7.2",
    "redux-form": "^7.2.3",
    "redux-thunk": "^2.2.0"
  },
  "devDependencies": {
    "babel-jest": "22.2.0",
    "babel-plugin-transform-decorators-legacy": "^1.3.4",
    "babel-plugin-transform-function-bind": "^6.22.0",
    "babel-preset-react-native": "4.0.0",
    "jest": "22.2.1",
    "react-test-renderer": "16.2.0"
  },
  "jest": {
    "preset": "react-native"
  }
}

```