react-native-camera/react-native-camera #1204: No barcode detection after taking photo

kruyvanna, 2018-02-02:
### Which implementation are you using

RCTCamera or RNCamera?
Both

### Steps to reproduce
1. onBarcodeRead, call camera.capture()
2. Point to other QR code

### Expected behaviour
onBarcodeRead should be still get called even after camera.capture()

### Actual behaviour
onBarcodeRead stop getting called

### Environment
- **Node.js version**: v8.9.1
- **React Native version**: 0.47.1
- **React Native platform + platform version**: iOS 9.0, Android 5.0

### react-native-camera
**Version**: ^0.10.0

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

niksajanjic, 2018-02-02:
For me, barcode scanning didn't work on version 0.12, only QR code scanning was successful. When I upgraded to 0.13 version, scanning of upc and ean barcodes. Didn't test other barcodes.

jgcmarins, 2018-02-02:
hello @kruyvanna, could you try on master branch please? thanks.
https://github.com/react-native-community/react-native-camera/blob/master/README.md#how-to-use-master-branch

kruyvanna, 2018-02-05:
Hi @jgcmarins I got compile error.
Looks like the package class has been moved?

[image]

kruyvanna, 2018-02-05:
It compiles now after I change the package to

` import org.reactnative.camera.RNCameraPackage;`

But the camera will show a static preview of the captured photo.
[without_start_preview.mp4.zip](https://github.com/react-native-community/react-native-camera/files/1693601/without_start_preview.mp4.zip)

I then figure out there is a **.startPreview()** I call it after calling **.capture()** and now it's working as expected. though only on current master.

jgcmarins, 2018-02-05:
nice to hear from you @kruyvanna
can we close this one?

kruyvanna, 2018-02-05:
@jgcmarins yes. i will do

kruyvanna, 2018-02-05:
thanks for helping on this issue

closed 2018-02-05 by kruyvanna

jgcmarins, 2018-02-05:
Thanks \o/
By the way, this info is already on [README.md](https://github.com/react-native-community/react-native-camera#android)

kruyvanna, 2018-02-05:
Ah, i see :D