react-native-camera/react-native-camera #1206: Camera.constants.BarCodeType is an empty object

niksajanjic, 2018-02-02:
### Expected behavior
Per documentation you should be able to pass an array of `barCodeTypes` using `Camera.constants.barCodeType` to allow only certain barcodes to be scanned.

### Actual behavior
Logging out `Camera.constants` object contains other constant objects, only `barCodeType` object is empty. That results in all codes scanned and passed to `onBarCodeRead` callback. The workaround is to directly pass expected strings depending on a Platform, see this issue https://github.com/react-native-community/react-native-camera/issues/652.

### Environment
- node 9.4
- RN 0.52.1
- Nexus 6 real device, Android 7.1.1

### react-native-camera
0.13

jgfidelis, 2018-02-02:
If you look the RCTCamera code here is the Android implementation for the constants getter:
```
private Map<String, Object> getBarCodeConstants() {
                return Collections.unmodifiableMap(new HashMap<String, Object>() {
                    {
                        // @TODO add barcode types
                    }
                });
            }```

This is a simple fix, there is a working version on RNCamera.
You can try using RNCamera in master branch.

niksajanjic, 2018-02-05:
@jgfidelis I tried your example and although it does fix a problem with constants (although in this example constants starts with a capital letter):
```
import { RNCamera } from 'react-native-camera'

console.log(RNCamera.Constants.BarCodeType)
```

There is a whole different problem. My app keeps crashing when trying to use `RNCamera` instead of `Camera`. Same code works fine when using `Camera`, but if I just try using `RNCamera`, my app will crash every time. The only error that seems probable cause that I found in Android Studio is:
```
02-05 15:19:59.077 11449-11449/com.glocaleze E/AndroidRuntime: FATAL EXCEPTION: main
                                                               Process: com.glocaleze, PID: 11449
                                                               java.lang.NoSuchMethodError: No static method getDisplay(Landroid/view/View;)Landroid/view/Display; in class Landroid/support/v4/view/ViewCompat; or its super classes (declaration of 'android.support.v4.view.ViewCompat' appears in /data/app/com.glocaleze-2/base.apk)
                                                                   at com.google.android.cameraview.CameraView.onAttachedToWindow(CameraView.java:136)
                                                                   at android.view.View.dispatchAttachedToWindow(View.java:15509)
                                                                   at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2916)
                                                                   at android.view.ViewGroup.addViewInner(ViewGroup.java:4456)
                                                                   at android.view.ViewGroup.addView(ViewGroup.java:4258)
                                                                   at com.facebook.react.views.view.ReactViewGroup.addView(ReactViewGroup.java:430)
                                                                   at android.view.ViewGroup.addView(ViewGroup.java:4198)
                                                                   at com.facebook.react.views.view.ReactViewManager.addView(ReactViewManager.java:247)
                                                                   at com.facebook.react.views.view.ReactViewManager.addView(ReactViewManager.java:38)
                                                                   at com.facebook.react.uimanager.NativeViewHierarchyManager.manageChildren(NativeViewHierarchyManager.java:401)
                                                                   at com.facebook.react.uimanager.UIViewOperationQueue$ManageChildrenOperation.execute(UIViewOperationQueue.java:180)
                                                                   at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:819)
                                                                   at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:926)
                                                                   at com.facebook.react.uimanager.UIViewOperationQueue.access$2100(UIViewOperationQueue.java:47)
                                                                   at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:986)
                                                                   at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
                                                                   at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:136)
                                                                   at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
                                                                   at android.view.Choreographer$CallbackRecord.run(Choreographer.java:872)
                                                                   at android.view.Choreographer.doCallbacks(Choreographer.java:686)
                                                                   at android.view.Choreographer.doFrame(Choreographer.java:618)
                                                                   at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:860)
                                                                   at android.os.Handler.handleCallback(Handler.java:751)
                                                                   at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                   at android.os.Looper.loop(Looper.java:154)
                                                                   at android.app.ActivityThread.main(ActivityThread.java:6119)
                                                                   at java.lang.reflect.Method.invoke(Native Method)
                                                                   at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
                                                                   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
```

I don't feel like opening an issue for something that is still in experimental phase, but I have been unable to use `RNCamera` and thus the issue with constants still persists for me at the moment.

sibelius, 2018-02-05:
@niksajanjic open an issue about it please

we are close to release RNCamera as the default camera component

jgfidelis, 2018-02-05:
@niksajanjic feel free to open an issue.

About the error, I am not very experienced with Android but it seems there is a mixture of Support app libraries between RNCamera and your project. The one that your project is using does not have the method that CameraView uses. Its a guess of mine. This thread may explain it better:
https://stackoverflow.com/questions/39049107/java-lang-nosuchmethoderror-no-static-method-setonapplywindowinsetslistener

Do you have any idea on how to fix this? Maybe change the support library package that you use on your app... may be a conflict between other react-native libraries as well.

akriger, 2018-02-15:
I had this same exact problem and the issue with me was that in my build.gradle dependencies in android/app/build.gradle  I had

compile 'com.android.support:appcompat-v7:23.0.1"

I changed the version to this:
compile 'com.android.support:appcompat-v7:26.+'

re-ran my app and everything is working

niksajanjic, 2018-02-16:
I will try upgrading to a newer version and I'll also try @akriger solution if needed and get back to you guys sometime next week. Thanks for the help.

imspikey, 2018-02-22:
I got the same problem after committing the support library in gradle file

Fconstant, 2018-02-27:
Is this still a problem?

niksajanjic, 2018-02-28:
I didn't have time to try upgrading to the new version so I really can't say.

closed 2018-02-28 by sibelius

Gmtas, 2018-07-04:
I have the same problem. Is there no information on how to solve it?

n1ru4l, 2018-12-19:
@Gmtas Please test it with the newest version of `RNCamera` and open a new issue if this error still occurs.