react-native-camera/react-native-camera #1207: CameraManager.Aspect on iOS

wassgha, 2018-02-02:
### Which implementation are you using

RNCamera
(only imported RNCamera through `import {RNCamera} from 'react-native-camera'`

[image]

sibelius, 2018-02-02:
This happen when the link is wrong

closed 2018-02-02 by wassgha

dann1609, 2018-03-19:
do a manual link:

1) npm install react-native-camera --save
2) In XCode, in the project navigator, right click Libraries ➜ Add Files to [your project's name]
3) Go to node_modules ➜ react-native-camera and add RNCamera.xcodeproj
4) In XCode, in the project navigator, select your project. Add libRNCamera.a to your project's Build Phases ➜ Link Binary With Libraries
5) Click RNCamera.xcodeproj in the project navigator and go the Build Settings tab. Make sure 'All' is toggled on (instead of 'Basic'). In the Search Paths section, look for Header Search Paths and make sure it contains both $(SRCROOT)/../../react-native/React and $(SRCROOT)/../../../React - mark both as recursive.

tiagoalucard, 2018-08-20:
expo bro! no need Mac

Dabbous1, 2018-09-24:
> do a manual link:
>
> 1. npm install react-native-camera --save
> 2. In XCode, in the project navigator, right click Libraries ➜ Add Files to [your project's name]
> 3. Go to node_modules ➜ [...]

What files do you add exactly?

n1ru4l, 2018-09-24:
@Dabbous1 The ones mentioned in Step 3