react-native-camera/react-native-camera #1214: Could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.

Fadi-Ghattas, 2018-02-05:
On android Bullied this what i get any help ? 

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApkCopy'.
   > Could not find com.github.react-native-community:cameraview:d5d9b0d925494ef451ce3eef3fdf14cc874d9baa.
     Required by:
         Kuppl:app:unspecified > Kuppl:react-native-camera:unspecified

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

sibelius, 2018-02-05:
follow https://github.com/react-native-community/react-native-camera#q--a

closed 2018-02-05 by sibelius

SteveDuqPro, 2018-02-15:
Is a bug of react-native-camera. You can't link it. So don't make a react-native link react-native-camera.

petrleocompel, 2018-02-16:
Updated link https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md

formigone, 2018-02-20:
What solved this for me was to follow [step #6 for Android](https://github.com/react-native-community/react-native-camera#getting-started):

    6) Add jitpack to android/build.gradle
    
    allprojects {
        repositories {
            maven { url "https://jitpack.io" }
        }
    }

ducpt2, 2018-02-22:
@formigone doest this lib work with RN version 0.53+? I try but it not work.

eroosenmaallen, 2018-02-23:
The QA docs linked above do not address this issue at all; there is no mention of this error.

akamahesh, 2018-04-04:
make sure you crosscheck with manual android instruction too. for installing automically

6) Add jitpack to android/build.gradle

allprojects {
    repositories {
        maven { url "https://jitpack.io" }
    }
}

did the trick for me!