react-native-camera/react-native-camera #1216: Different argument structure of event onBarCodeRead on iOS and Android

anchoretics, 2018-02-06:
### Which implementation are you using

RCTCamera or RNCamera?
RNCamera
### Steps to reproduce
1.barCodeTypes={[Camera.constants.BarCodeType.qr]}
2.onBarCodeRead={this.onBarCodeRead.bind(this)}
3.Scan a QR Code on iOS and android device

### Expected behaviour
the argument of event onBarCodeRead, should be the structure as follows:
```
{
    data: "https://xxx.xxx",
    type: "QR_CODE",
    bounds: {
        origin: {x:0, y:0},
        size: {height: 0, width: 0}
    }
}
```

### Actual behaviour
the onBarCodeRead's argument is expected on iOS,
but it's anothor structure on android device:
```
{
    data: "https://xxx.xxx",
    type: "QR_CODE",
    bounds: [
        {x:0, y:0},
        {x:0, y:0},
        {x:0, y:0},
        {x:0, y:0}
    ]
}
```
and there's no related document about this,
is this a BUG?
whatever, there shoud be a detailed description aboud onBarCodeRead and it's arguments.

### Environment
- **Node.js version**: 8.9.3
- **React Native version**: 0.52.0
- **React Native platform + platform version**: iOS 11.2, Android 5.1

### react-native-camera
**Version**: 0.13.0

jgfidelis, 2018-02-06:
Are you on master or v0.13.0? You cannot use RNCamera on 0.13. You must be using RCTCamera then.

anchoretics, 2018-02-07:
@jgfidelis tks, i changed to RNCamera, bu onBarCodeRead's argument donot have bounds anymore,
I just need the bounds to check that if the scaned QRcode is inside the area of my customer mark as blow:
[image]

n1ru4l, 2018-11-20:
Seems like a feature request for RNCamera! Feel free to send a pull request that implements the bounds for `RNCamera`

SimonErm, 2018-11-20:
@n1ru4l  Yes, the actual behaviour is the expected behaviour. Like said in the discription of  #1402 (@sash ) , Android and iOS returns different bounds since ZXing returns no real bounds, but ResultPoints (https://zxing.github.io/zxing/apidocs/com/google/zxing/ResultPoint.html). I though that this already was added to the docs, so i will make a PR.

SimonErm, 2018-11-20:
@anchoretics which version of react-native-camera are you using?

n1ru4l, 2018-11-20:
@SimonErm Can't we convert the android bounds to the ios equivalent?

SimonErm, 2018-11-20:
@n1ru4l  We could test, which codes deliver which resultPoints and calculate the origin and size before emitting the event. The bounds wouldn't be exact since the resultPoints don't deliver data to calculate the barcode heights and the QR-Corners are not the outer corners, but i think it would be more convenient for the user to handle iOS and Android the same way.

SimonErm, 2018-11-20:
I will try to do this at the weekend

SimonErm, 2018-11-25:
I worked on it, but i don't think this is the solution, because the 2D codes returning different ResultPoints

closed 2018-12-18 by anchoretics