react-native-camera/react-native-camera #1218: RCTCamera  App not working in android 4.4.4

ChandraIOT, 2018-02-07:
We have created an app which is using RCTCamera  package, app is working fine on android above then 4.4.4 but not working in this version. When we click on app icon app is not launching.

jgfidelis, 2018-02-07:
Have you tried master branch?

ChandraIOT, 2018-02-07:
We tried from master branch but when we installing node module dependency by npm install under example directory it is giving us attached error

[image]

ChandraIOT, 2018-02-07:
Also tried using master branch in our app by given "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera" in package.json file but still same issue is coming.

ChandraIOT, 2018-02-09:
Getting below crash logs on app launch.

Fatal Exception: java.lang.VerifyError: org/reactnative/camera/CameraViewManager
       at com.lwansbrough.RCTCamera.RCTCameraPackage.createViewManagers(RCTCameraPackage.java:35)
       at com.facebook.react.ReactInstanceManager.createAllViewManagers(ReactInstanceManager.java:715)
       at com.facebook.react.CoreModulesPackage.createUIManager(CoreModulesPackage.java:198)
       at com.facebook.react.CoreModulesPackage.access$200(CoreModulesPackage.java:73)
       at com.facebook.react.CoreModulesPackage$8.get(CoreModulesPackage.java:150)
       at com.facebook.react.CoreModulesPackage$8.get(CoreModulesPackage.java:147)
       at com.facebook.react.LazyReactPackage.createNativeModules(LazyReactPackage.java:81)
       at com.facebook.react.NativeModuleRegistryBuilder.processPackage(NativeModuleRegistryBuilder.java:106)
       at com.facebook.react.ReactInstanceManager.processPackage(ReactInstanceManager.java:1071)
       at com.facebook.react.ReactInstanceManager.processPackages(ReactInstanceManager.java:1042)
       at com.facebook.react.ReactInstanceManager.createReactContext(ReactInstanceManager.java:980)
       at com.facebook.react.ReactInstanceManager.access$600(ReactInstanceManager.java:109)
       at com.facebook.react.ReactInstanceManager$4.run(ReactInstanceManager.java:802)
       at java.lang.Thread.run(Thread.java:841)

badpenguin, 2018-02-14:
same here with 4.4.2 :-( will try master branch or upgrade to droid-5

gaboelnuevo, 2018-02-21:
Same Problem?

E/memtrack(23137): Couldn't load memtrack module (No such file or directory)
E/android.os.Debug(23137): failed to load memtrack module: -2
E/AndroidRuntime(23147): FATAL EXCEPTION: Thread-1291
E/AndroidRuntime(23147): Process: com.myapp.app, PID: 23147
E/AndroidRuntime(23147): java.lang.VerifyError: org/reactnative/camera/CameraViewManager
E/AndroidRuntime(23147): 	at org.reactnative.camera.RNCameraPackage.createViewManagers(RNCameraPackage.java:40)
E/AndroidRuntime(23147): 	at com.facebook.react.ReactInstanceManager.createAllViewManagers(ReactInstanceManager.java:792)
E/AndroidRuntime(23147): 	at com.facebook.react.CoreModulesPackage.createUIManager(CoreModulesPackage.java:175)
E/AndroidRuntime(23147): 	at com.facebook.react.CoreModulesPackage.access$200(CoreModulesPackage.java:55)
E/AndroidRuntime(23147): 	at com.facebook.react.CoreModulesPackage$7.get(CoreModulesPackage.java:132)
E/AndroidRuntime(23147): 	at com.facebook.react.CoreModulesPackage$7.get(CoreModulesPackage.java:129)
E/AndroidRuntime(23147): 	at com.facebook.react.LazyReactPackage.createNativeModules(LazyReactPackage.java:81)
E/AndroidRuntime(23147): 	at com.facebook.react.NativeModuleRegistryBuilder.processPackage(NativeModuleRegistryBuilder.java:106)
E/AndroidRuntime(23147): 	at com.facebook.react.ReactInstanceManager.processPackage(ReactInstanceManager.java:1202)
E/AndroidRuntime(23147): 	at com.facebook.react.ReactInstanceManager.processPackages(ReactInstanceManager.java:1172)
E/AndroidRuntime(23147): 	at com.facebook.react.ReactInstanceManager.createReactContext(ReactInstanceManager.java:1107)
E/AndroidRuntime(23147): 	at com.facebook.react.ReactInstanceManager.access$900(ReactInstanceManager.java:113)
E/AndroidRuntime(23147): 	at com.facebook.react.ReactInstanceManager$5.run(ReactInstanceManager.java:944)
E/AndroidRuntime(23147): 	at java.lang.Thread.run(Thread.java:841)

matthiasleitner, 2018-02-22:
Seeing the same issue with 4.4.2 using master. @jgfidelis any suggestions how to best investigate this issue?

closed 2019-04-18 by sibelius