react-native-camera/react-native-camera #1220: takePictureAsync  returns error Android

Darwex, 2018-02-07:
### Which implementation are you using

RNCamera?

### Steps to reproduce
1. Install camera according to docs
2. Run on android 
3. Call takePictureAsync 

### Expected behaviour
Camera should take i picture, it is working for me on iOS device.

### Actual behaviour
When calling takePictureAsync() i receive following error in catch 
```
Error: takePictureAsync: Expected a Camera componen
```
relevant part of code

```
import {RNCamera} from 'react-native-camera'
...

  handleCapturePhoto = async() => {
    try {
     // adding activity indicator because of low performance on Android with previous version
      await this.setState({isCapturing: true})

      const cameraData = await this.camera.takePictureAsync()
      this.setState({
        takenPhoto: cameraData.uri,
        isCapturing: false
      })
    } catch (e) {
     // This logs the error
      console.log(e)
    }
  }

....

// Called in component render
renderCamera() {
      <RNCamera
        ref={ref => this.camera = ref}
        style={styles.camera}
        type={this.state.cameraType}
      >
        {this.renderActivityIndicator()}
        {goBackButton}
        {capturePhotoButton}
        {switchCameraButton}
      </RNCamera>
}

```

### Environment
- **Node.js version**: 8.6.0
- **React Native version**: 0.52.0
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**: master

jgfidelis, 2018-02-08:
I think I had similar problem. Can you try not using any setState inside your takePictureAsync() method? See if it helps.

Darwex, 2018-02-08:
Thanks for the tip. I just tried it but then I get this error 

[image]

jgfidelis, 2018-02-09:
Do you know native development? You could try putting breakpoints and logging in the Java code to see where is happening.

Your code looks ok..
```
handleCapturePhoto = async() => {
    try {
      const cameraData = await this.camera.takePictureAsync()
      console.log(cameraData.uri);
    } catch (e) {
     // This logs the error
      console.log(e)
    }
  }
```
Should work fine. The problem would be in the native code :/

jgfidelis, 2018-02-09:
You may have more than one RNCamera component instantiated in your view hierarchy. See if your onMountError function is being called at anytime. Android does not let you have more than one RNCameraView at the same time. If you are on a screen that has an RNCamera, make sure to pop that screen from the stack before opening another and mounting another RNCamera.

Darwex, 2018-02-11:
Cool thanks for the hints. I will try that and let you know. I checked the native code and it seems that it might be issue with viewTag so it corresponds with your multiple cameras mounted theory.

Darwex, 2018-02-11:
FYI `onMountError` is never called.

Darwex, 2018-02-11:
When i pass catched error form Java to RN logger I can see this 

```
Error: Unknown pattern character 'Y'
    at createErrorFromErrorData (NativeModules.js:123)
    at NativeModules.js:80
    at MessageQueue.__invokeCallback (MessageQueue.js:400)
    at MessageQueue.js:139
    at MessageQueue.__guardSafe (MessageQueue.js:316)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (MessageQueue.js:138)
```

Catch is located in takePicture method inside CameraModule.java line 289.

jgfidelis, 2018-02-12:
https://stackoverflow.com/questions/16646124/java-dateformat-illegal-pattern-character-y

This error looks like something when creating the photo and adding the date, what language is your device on? Your device language may use a different character other than 'Y' for year and this confuses android.

It seems like a bug on CameraView.

Could you set your device's region and language to US/English?

Darwex, 2018-02-12:
Yep i tried Googling that as well. Came to the same conclusion. I am running it on simulator and that has everything set to US english already.

sibelius, 2018-02-12:
@Darwex can u provide the native stack trace?

jgfidelis, 2018-02-12:
Do you have a device to test on? Simulator has some weird behavior, it seems

Darwex, 2018-02-14:
It seems it behaves the same on real device as well 😢 

The stacktrace:

```
Error: java.lang.IllegalArgumentException: Unknown pattern character 'Y'
	at java.text.SimpleDateFormat.validatePatternCharacter(SimpleDateFormat.java:323)
	at java.text.SimpleDateFormat.validatePattern(SimpleDateFormat.java:305)
	at java.text.SimpleDateFormat.<init>(SimpleDateFormat.java:365)
	at org.reactnative.camera.RNCameraViewHelper.generateSimulatorPhoto(RNCameraViewHelper.java:295)
	at org.reactnative.camera.CameraModule$3.execute(CameraModule.java:205)
	at com.facebook.react.uimanager.UIViewOperationQueue$UIBlockOperation.execute(UIViewOperationQueue.java:510)
	at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:819)
	at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:926)
	at com.facebook.react.uimanager.UIViewOperationQueue.access$2100(UIViewOperationQueue.java:47)
	at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:986)
	at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
	at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:136)
	at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:856)
	at android.view.Choreographer.doCallbacks(Choreographer.java:670)
	at android.view.Choreographer.doFrame(Choreographer.java:603)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:844)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:148)
	at android.app.ActivityThread.main(ActivityThread.java:5417)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

	at java.text.SimpleDateFormat.validatePatternCharacter(SimpleDateFormat.java:323)
	at java.text.SimpleDateFormat.validatePattern(SimpleDateFormat.java:305)
	at java.text.SimpleDateFormat.<init>(SimpleDateFormat.java:365)
	at org.reactnative.camera.RNCameraViewHelper.generateSimulatorPhoto(RNCameraViewHelper.java:295)
	at org.reactnative.camera.CameraModule$3.execute(CameraModule.java:205)
	at com.facebook.react.uimanager.UIViewOperationQueue$UIBlockOperation.execute(UIViewOperationQueue.java:510)
	at com.facebook.react.uimanager.UIViewOperationQueue$1.run(UIViewOperationQueue.java:819)
	at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:926)
	at com.facebook.react.uimanager.UIViewOperationQueue.access$2100(UIViewOperationQueue.java:47)
	at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:986)
	at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:31)
	at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:136)
	at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:107)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:856)
	at android.view.Choreographer.doCallbacks(Choreographer.java:670)
	at android.view.Choreographer.doFrame(Choreographer.java:603)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:844)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:148)
	at android.app.ActivityThread.main(ActivityThread.java:5417)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

    at createErrorFromErrorData (blob:http://localhost:8081/9b42fc4b-f231-44b7-ac96-d7e4d104abeb:3617:17)
    at blob:http://localhost:8081/9b42fc4b-f231-44b7-ac96-d7e4d104abeb:3569:27
    at MessageQueue.__invokeCallback (blob:http://localhost:8081/9b42fc4b-f231-44b7-ac96-d7e4d104abeb:4011:18)
    at blob:http://localhost:8081/9b42fc4b-f231-44b7-ac96-d7e4d104abeb:3756:18
    at MessageQueue.__guardSafe (blob:http://localhost:8081/9b42fc4b-f231-44b7-ac96-d7e4d104abeb:3924:11)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (blob:http://localhost:8081/9b42fc4b-f231-44b7-ac96-d7e4d104abeb:3755:14)
    at http://localhost:8081/debugger-ui/debuggerWorker.js:72:58
```

jgfidelis, 2018-02-15:
@Darwex On RNCameraViewHelper, cna you change the line 295 from:
```
SimpleDateFormat simpleDateFormat = new SimpleDateFormat("dd.MM.YY HH:mm:ss", Locale.getDefault());
```

to

```
SimpleDateFormat simpleDateFormat = new SimpleDateFormat("dd.MM.YY HH:mm:ss", Locale.US);
```

Thanks.

Darwex, 2018-02-19:
I did the change... And sadly I am back to the error we started with 😢 

```
Error: takePictureAsync: Expected a Camera component
    at createErrorFromErrorData (NativeModules.js:123)
    at NativeModules.js:80
    at MessageQueue.__invokeCallback (MessageQueue.js:400)
    at MessageQueue.js:139
    at MessageQueue.__guardSafe (MessageQueue.js:316)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (MessageQueue.js:138)
    at debuggerWorker.js:72
```

Darwex, 2018-02-19:
I tried that, unfortunately no change.

Darwex, 2018-02-19:
I also tried changing drawText method (in order to skip usage of date) in `generateSimulatorPhoto` to 
```
    canvas.drawText("photo", width * 0.1f, height * 0.9f, textPaint);
```

But then app crashes immediately with no error message.

ratkus91, 2018-02-21:
# Crashlytics - plaintext stacktrace downloaded by Andrii Sementsov at Wed, 21 Feb 2018 17:36:18 GMT
# URL: https://fabric.io/ourcnc-oy/android/apps/com.ourcncreactnative/issues/5a8d46608cb3c2fa6380ce03/sessions/latest
# Organization: OurCnC Oy
# Platform: android
# Application: OurCnCReactNative
# Version: 1.0.11 (1)
# Bundle Identifier: com.ourcncreactnative
# Issue #: 15
# Issue ID: 5a8d46608cb3c2fa6380ce03
# Session ID: 5A8DAA70031000012BC631D5AB1AC6B3_DNE_0_v2
# Date: 2018-02-21T17:21:00Z
# OS Version: 6.0
# Device: Android SDK built for x86_64
# RAM Free: 71.5%
# Disk Free: 81.1%

#0. Crashed: AsyncTask #3
       at android.os.AsyncTask$3.done(AsyncTask.java:309)
       at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
       at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
       at java.util.concurrent.FutureTask.run(FutureTask.java:242)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

--

Fatal Exception: java.lang.RuntimeException: An error occurred while executing doInBackground()
       at android.os.AsyncTask$3.done(AsyncTask.java:309)
       at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
       at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
       at java.util.concurrent.FutureTask.run(FutureTask.java:242)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

Caused by java.lang.NoClassDefFoundError: Failed resolution of: Landroid/support/media/ExifInterface;
       at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:64)
       at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:26)
       at android.os.AsyncTask$2.call(AsyncTask.java:295)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

Caused by java.lang.ClassNotFoundException: Didn't find class "android.support.media.ExifInterface" on path: DexPathList[[zip file "/data/app/com.ourcncreactnative-1/base.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_dependencies_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_0_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_1_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_2_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_3_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_4_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_5_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_6_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_7_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_8_apk.apk", zip file "/data/app/com.ourcncreactnative-1/split_lib_slice_9_apk.apk"],nativeLibraryDirectories=[/data/app/com.ourcncreactnative-1/lib/x86, /data/app/com.ourcncreactnative-1/base.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_dependencies_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_0_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_1_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_2_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_3_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_4_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_5_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_6_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_7_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_8_apk.apk!/lib/x86, /data/app/com.ourcncreactnative-1/split_lib_slice_9_apk.apk!/lib/x86, /vendor/lib, /system/lib]]
       at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
       at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
       at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
       at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:64)
       at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:26)
       at android.os.AsyncTask$2.call(AsyncTask.java:295)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

#0. Crashed: AsyncTask #3
[... cut ...]
       at okhttp3.internal.ws.RealWebSocket.loopReader(RealWebSocket.java:262)
       at okhttp3.internal.ws.RealWebSocket$2.onResponse(RealWebSocket.java:201)
       at okhttp3.RealCall$AsyncCall.execute(RealCall.java:135)
       at okhttp3.internal.NamedRunnable.run(NamedRunnable.java:32)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

sibelius, 2018-02-21:
@ratkus91 missing exif package

ratkus91, 2018-02-21:
@sibelius Thanks.
Next error:
# Crashlytics - plaintext stacktrace

#0. Crashed: AsyncTask #3
       at android.os.AsyncTask$3.done(AsyncTask.java:309)
       at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
       at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
       at java.util.concurrent.FutureTask.run(FutureTask.java:242)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

--

Fatal Exception: java.lang.RuntimeException: An error occurred while executing doInBackground()
       at android.os.AsyncTask$3.done(AsyncTask.java:309)
       at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
       at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
       at java.util.concurrent.FutureTask.run(FutureTask.java:242)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'int android.graphics.Bitmap.getWidth()' on a null object reference
       at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:84)
       at org.reactnative.camera.tasks.ResolveTakenPictureAsyncTask.doInBackground(ResolveTakenPictureAsyncTask.java:26)
       at android.os.AsyncTask$2.call(AsyncTask.java:295)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

#0. Crashed: AsyncTask #3
       at android.os.AsyncTask$3.done(AsyncTask.java:309)
       at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
       at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
       at java.util.concurrent.FutureTask.run(FutureTask.java:242)
       at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

ratkus91, 2018-02-22:
Today I found one interesting case.
On the android camera works if you enable "Debug JS Remotely".

SamMatthewsIsACommonName, 2018-02-23:
I have a weird situation where if I update the UI, the async operation never returns. For example if I try and show a loading indicator while it is capturing the photo. I tried avoiding using setState by going from props (redux), as well as having a nested component with its own state (which would set to loading etc...), and both lead to same outcome. The camera screen would go black and the function would never return. Just thought I'd add that in for info thanks!

ospfranco, 2018-03-02:
Sorry, miss click!

Darwex, 2018-03-12:
Hey Guys, any update? Any help I can provide? 
Thanks.

brunolemos, 2018-03-12:
Just tried upgrading from RCTCamera to RNCamera. 
Unfortunately I have the same crash reports as @ratkus91.

(Related: #1340)

bang88, 2018-03-13:
Its happened in emulator but our device seems work well. 😢

Darwex, 2018-03-13:
Not good 😀 So can I provide any additional information or try something else? ☺️

Darwex, 2018-03-13:
BTW is there any fix coming for the simulator?

cl-will, 2018-03-15:
There seems to be two errors going on:

- date format error
- stream decoding error

I got it working using the following patch.  I'm going to create a PR shortly as well:

[react-native-camera+1.0.2.patch.txt](https://github.com/react-native-community/react-native-camera/files/1817308/react-native-camera.1.0.2.patch.txt)

Narutuffy, 2018-03-17:
@cl-will It still shows the same error for me after I had updated it with the patch..

closed 2018-03-18 by jgfidelis

KavithaTravash, 2018-03-19:
I'm also getting the same issue, is this issue fixed? please help me out.

cl-will, 2018-03-19:
@Narutuffy Can you add some logging to the takePicture method and let me know what it says?  Something like:

```java
          try {
              if (!Build.FINGERPRINT.contains("generic")) {
                if (cameraView.isCameraOpened()) {
                  cameraView.takePicture(options, promise, cacheDirectory);
                } else {
                  promise.reject("E_CAMERA_UNAVAILABLE", "Camera is not running");
                }
              } else {
                  Bitmap image = RNCameraViewHelper.generateSimulatorPhoto(cameraView.getWidth(), cameraView.getHeight());

                  // added line
                  android.util.Log.i("ReactNativeJS", "takePhoto image: " + image);

                  ByteArrayOutputStream stream = new ByteArrayOutputStream();
                  image.compress(Bitmap.CompressFormat.JPEG, 100, stream);
                  byte[] byteArray = stream.toByteArray();

                  // added line
                  android.util.Log.i("ReactNativeJS", "takePhoto byteArray: " + byteArray);

                  new ResolveTakenPictureAsyncTask(byteArray, promise, options, cacheDirectory).execute();
              }
        } catch (Exception e) {
          // added line
          android.util.Log.i("ReactNativeJS", "takePhoto error: " + e.getMessage());
          promise.reject("E_CAMERA_BAD_VIEWTAG", "takePictureAsync: Expected a Camera component");
        }
```

Then run `react-native log-android | grep takePhoto` and let me know what the output is when you try to take a picture?

nlothian, 2018-03-27:
I'm getting this problem too on a real device.

My Javascript looks like this:

```
        console.log("About to takePictureAsync")
        const data = await this.camera.takePictureAsync(options)
        console.log("takePictureAsync done")
```
I've debugged and modified the native takePicture method to have more logging:

```
                if (cameraView.isCameraOpened()) {
                  android.util.Log.i("ReactNativeJS", "Before cameraView.takePicture");
                  cameraView.takePicture(options, promise, cacheDirectory);
                  android.util.Log.i("ReactNativeJS", "After cameraView.takePicture");
                } else {
```
My log looks like this:
```

I/ReactNativeJS( 7624): About to takePictureAsync
I/ReactNativeJS( 7624): Before cameraView.takePicture
I/ReactNativeJS( 7624): After cameraView.takePicture

```

It never hits the "console.log("takePictureAsync done")" line in my JS code. When I debug it drops out of the UIBlock.execute method, and I'm not sure where it goes after that. 

Happy to help debug, but I don't really know React or Android so I'll need some hand holding, sorry!

nlothian, 2018-03-31:
Note that the reason the date format was failing for some people is that the "YY" string only became available in API version 24. "yyyy" (which @cl-will's patch uses) is more compatible. This was only a problem on the emulator though. 

I'm still having trouble on physical devices.

luisarmandodv, 2018-05-14:
Same problem here, I need to add a loader when a picture is taken, but a got exactly what  @SamMatthewsIsACommonName describes. Any update?

luisarmandodv, 2018-05-14:
I resolved this problem adding the loader component outside RNCamera component and take everything inside an <View>, now it works in both android and iOS devices. 

> <View style={{ flex: 1 }}>
      <View style={{ flex: 1 }}>
        <RNCamera
          ref={cam => {
            this.camera = cam;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
        >
            <TouchableOpacity
              disabled={this.props.loading}
              onPress={this.takePicture.bind(this)}
            >
                <Text>Press</Text>
            </TouchableOpacity>
        </RNCamera>
        {this.getActivityIndicator()}
      </View>

> <View style={{ flex: 1 }}>
```
  takePicture = async function() {
    if (this.camera) {
      const options = { quality: 0.5, base64: true };
      this.setState({
        loading: true
      });
      const data = await this.camera.takePictureAsync(options);
      this.setState({
        loading: false
      });
    }
  };

Gatreek92, 2018-08-03:
Hi guys, does anyone tried`onPictureSaved` function ? (Its a callback that you can pass to `takePictureAsync` and it will be involved when the picture is saved, it came with **SDK 28**)
see doc [here](https://github.com/expo/expo-docs/blob/9d5f0e478d1f13a67546e71928b5ebbc61a86799/versions/unversioned/sdk/camera.md#takepictureasync)

MrIceman, 2019-01-10:
any news to this issue?

skizzo, 2019-03-07:
I'm facing the same problem, this is supposed to be pretty standard functionality.. Any updates on this?

jahicDario, 2019-04-03:
Same problem here: 

> Error: takePictureAsync: Expected a Camera component
    at createErrorFromErrorData (blob:file:///36e0ebee-931c-42d5-a284-73487a92e556:2023:17)
    at blob:file:///36e0ebee-931c-42d5-a284-73487a92e556:1975:27
    at MessageQueue.__invokeCallback (blob:file:///36e0ebee-931c-42d5-a284-73487a92e556:2526:18)
    at blob:file:///36e0ebee-931c-42d5-a284-73487a92e556:2273:18
    at MessageQueue.__guard (blob:file:///36e0ebee-931c-42d5-a284-73487a92e556:2430:13)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (blob:file:///36e0ebee-931c-42d5-a284-73487a92e556:2272:14)
    at t (file:///home/dario/ReactJS/rn-debugger-linux-x64/resources/app.asar/js/RNDebuggerWorker.js:1:42788)

ghasemikasra39, 2019-05-25:
same issue. any solution?

franck-nadeau, 2019-07-23:
I had the same issue, worked fine on iOS but didn't on Android. I found out that I had I had two problems.

1. First off the this is not properly bounded on Android. Not sure how that is possible, but anyway I had to change my code as follows:

````
    <RNCamera
            ....
            onBarCodeRead={this.onBarCodeRead.bind(this)}
          />

````

To this:

````
    <RNCamera
            ....
            onBarCodeRead={event => {
              this.onBarCodeRead(event, this);
            }}
          />
````
and use the passed object instead of this inside my ````onBarCodeRead```` function. i.e. ````onBarCodeRead = (event, that) => {  that.camera.takePictureAsync( ````

2. I threw in some debugs in the Java code and found out that Android's implementation does not allow me to pause the video before taking an image while iOS's does (see error message in the image below.)

[image]

So I removed the ````pausePreview()```` that I had before the ````takePictureAsync()```` call , and it worked.

jackkrone, 2023-04-27:
> Its happened in emulator but our device seems work well. 😢

For those of you encountering the issue only on the emulator, it may be related to the camera settings of your AVD (android virtual device). Specifically, there is an issue when using "VirtualScene" or "Webcam0". See here: https://github.com/react-native-camera/react-native-camera/issues/3141#issuecomment-1525911195

PR #1356 (2018-03-18): Fix stream error and date format error
## Fix date formatting error

There was an error with the date format, this new text string seems to work without error.

## Fix bitmap stream error

The ResolveTakenPictureAsyncTask doInBackground method was unable to decode the byte array that it was given by RNCameraViewHelper.generateSimulatorPhoto.  The decoding always returned null, which caused the app to crash when trying to getWidth on it.

This creates a byte array of the expected type (jpeg) and passes it to the ResolveTakenPictureAsyncTask constructor

Fixes #1220
commit 17eebff4 (2018-03-18): Fix stream error and date format error

Fixes #1220