react-native-camera/react-native-camera #1221: new RCTCameraPackage() appears two times inside getPackages()

sushantsaho, 2018-02-08:
### Which implementation are you using

RCTCamera or RNCamera?
RCTCamera

### Steps to reproduce
1. rm -rf node_modules/
2. npm install
3. react-native link

Repeat above steps again.

### Expected behaviour
Tell us what should happen
new RCTCameraPackage() should appear once inside getPackages() method
### Actual behaviour
Tell us what happens instead
new RCTCameraPackage() appears two times inside getPackages() method also two import statement appears
### Environment
- **Node.js version**: v9.4.0
- **React Native version**: 0.51.0
- **React Native platform + platform version**:  Android 8.0

### react-native-camera
**Version**: npm version or "master"(Happening in both the npm version)

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-02-08:
I do not understand, if you linked once, why will you link it again? Once you did it once and added those lines to your App project, you do need to do it again. You can then just `npm install`.

sushantsaho, 2018-02-08:
when i add any other library and it has native dependancy, i need to do react-native link, that time it will add again.

sibelius, 2018-02-08:
This is a problem with React native link script

closed 2018-02-08 by sibelius

sushantsaho, 2018-02-08:
But why its not happening with other library

sushantsaho, 2018-02-08:
Its happening after adding the below code inside build.gradle

compile (project(':react-native-camera')) {
      exclude group: "com.android.support"
    }
    compile ('com.android.support:exifinterface:26.0.1') {
        force = true;
    }

SirVeryBritish, 2018-05-01:
We are experiencing the same issue with react-native-camera 1.0.3 on react-native 0.51.0 but only on Android, iOS works fine. Is it possible to reopen this issue? I'm not aware that this happens with other packages.