react-native-camera/react-native-camera #1222: Camera.constants.Orientation.landscapeLeft  in  iOS

Jinran, 2018-02-08:
when I use orientation in ios
orientation={Camera.constants.Orientation.landscapeLeft}
camera set to left 
but android camera is the correct direction and return the photo is landscapeLeft

how can I use camera with correct direction and return photo in my settings way, in some OCR scene
such as vertical screen take photo  return a landscape left photo in ios

### Environment
- **Node.js version**: 7.5.0
- **React Native version**: 0.53.0
- **React Native platform + platform version**: iOS 11.2, Android 6.0, etc

### react-native-camera
**Version**: npm version or "master"

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate

jgfidelis, 2018-02-08:
Can you try with RNCamera? Inside your package.json, use this "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera" instead of "react-native-camera": "^0.12.0"

Check the migration and RNCamera doc in https://github.com/react-native-community/react-native-camera/tree/master/docs

Fconstant, 2018-02-27:
You can use [react-native-orientation](https://github.com/yamill/react-native-orientation).