react-native-camera/react-native-camera #1229: Removed props in new RNCamera

Deevent, 2018-02-12:
Hello,

I just successfully switched to RNCamera on my project but it appears that some features have disappeared in the new version (1.0.0).

aspect={RNCamera.Constants.Aspect.fill}
captureTarget={RNCamera.Constants.CaptureTarget.disk}

I was using those two properties on the Camera component but I can't anymore. I want my camera to be fullscreen and for that, I need to set the aspect. I went through the code and it doesn't exist anymore.

jgfidelis, 2018-02-12:
You can keep using RCTCamera if RNCamera still does not satisfy your needs.

I did not fill any need of the aspect prop in RNCamera.

RNCamera is a whole new implementation of the camera in relation to RCTCamera, so some rctcamera feature are still not present. But RNCamera is way faster.

piuccio, 2018-03-26:
I just had the same issue, it works fine without `aspect` but I really miss the `captureTarget`. I'm using `react-native-fetch-blob` to upload recorded videos to my server but I don't get access to a physical file anymore.

I don't want to use `fetch` because it doesn't have progress information

n1ru4l, 2018-09-25:
@Deevent Feel free to open a pull request that reimplements those properties.

@piuccio wrote
> I just had the same issue, it works fine without `aspect` but I really miss the `captureTarget`. I'm using `react-native-fetch-blob` to upload recorded videos to my server but I don't get access to a physical file anymore.
> 
> I don't want to use `fetch` because it doesn't have progress information

recordAsync returns the uri of the file, so you are still able to get physical access to the file.

closed 2018-10-05 by Deevent

bcalik, 2018-10-24:
@n1ru4l ok we can access the file from cache, but when you make an application that has offline capabilities, then cache directory is not reliable because I have to keep the recordings for later use.

n1ru4l, 2018-10-24:
@bcalik feel free to copy it to another place or send a pull request that allows specifying the output path of the file

rodrigoreis22, 2019-03-03:
Does anyone have an alternative for the missing `captureTarget` using `react-native-fetch-blob` ? I'm also using it to upload to my server with a progress bar.

piuccio, 2019-03-03:
@rodrigoreis22 I'm using it without `captureTarget` and with `react-native-fetch-blob` and it works like this:

```js
import fetchBlob from 'rn-fetch-blob';

const stripFilePathPrefix = (pathStr) => {
  if (pathStr.startsWith('file://')) return pathStr.replace(/^file:\/\//, '');
  return pathStr;
};

class MyCamera extends React.Component() {
  onTouchStartRecording() {
    this.camera.recordAsync().then((data) => {
      const videoPath = data.uri;
      const header = {}; // maybe you need authentication here
      fetchBlob.fetch('PUT', 'your-url', headers, fetchBlob.wrap(stripFilePathPrefix(payload))).then(() => {});
    });
  }
  render () {}
}
```

Keep in mind that `react-native-fetch-blob` is not maintained anymore, `rn-fetch-blob` is a fork with a few bug fixes, but it's not well maintained either, there might be better ways now...

rodrigoreis22, 2019-03-03:
Thanks @piuccio . I'll try this implementation.